foxglove/studio #2795: table export to CSV

nirhenn, 2022-02-07:
I have a ros topic where I would like to export all messages with all attributes to a CSV (no filters). It will be nice if I can use a table pan with a topic and export it to a CSV of a full recording. Ofcure an improve subset can include filters but min requirements is a full topic

defunctzombie, 2022-02-07:
@nirhenn what do messages on the topic look like (i.e. what are their fields)? and what would you expect the output CSV to look like for these messages?

nirhenn, 2022-02-09:
I am recoding standard CANBUS messages which I parsed using DBC “dictionary”.I would like to have a csv file where the columns contains is ID and related value in a times series per line. Hope it helps, will be more than happy to jump on a call to do some more thinking. Nir Sent from Mail for Windows