foxglove/studio #2796: Add `datatype` field to MessageEvent

defunctzombie, 2022-02-07:
When a panel receives message events for subscribed topics, it might need to take different code paths depending on the datatype of the received message.

For example - the Image panel needs to know if the message it received should be processed as a ROS `ImageMarkerArray` or a [`foxglove.ImageAnnotations`](https://github.com/foxglove/website/pull/334) message. The datatype name determines which fields are expected and their semantics.

The existing array of message events does not convey this information as only ROS message datatypes were previously supported.

amacneil, 2022-02-08:
We use `schemaName` in both mcap and ws-protocol, so I vote we start using that term consistently everywhere, and clean up places in our UI (and preferably our codebase too) where other terms like `datatype` are used indiscriminately.

defunctzombie, 2022-02-08:
The consensus is `schemaName` is a better term. I've updated the issue title.

jtbandes, 2022-10-07:
Done in #4576

closed 2022-10-07 by jtbandes

defunctzombie, 2022-10-07:
👏

amacneil, 2022-10-07:
I have no memory of making this comment but apparently I am internally consistent over 6 months.