foxglove/studio #2797: Draw trajectory of GPS positions using Node Playground

kis-balazs, 2022-02-07:
Since the current implementation of the Node Playground using **Messages.sensor_msgs__NavSatFix** when transforming from a non-standard GPS coordinate into a standardised form allows the user to plot every point one after the other without the possibility of locking existing points, is there a way to keep GPS coordinates on the map without publishing them, or having them previously in the ROS bag?

kis-balazs, 2022-02-07:
Hello!
Let me create a gist shortly :)
https://gist.github.com/kis-balazs/9fcd0e95f21422777e9a88fe15d240e5

kis-balazs, 2022-02-07:
Oh, so that means it cannot keep a 'trajectory' based on points drawn from Node Playground, is it?

defunctzombie, 2022-02-07:
> Oh, so that means it cannot keep a 'trajectory' based on points drawn from Node Playground, is it?

The panel could be made to do whatever - but today it does not keep any such history. The plotting of other points you see (when viewing a bag file), is from a different mechanism that has "preloaded" those points. That mechanism is separate from the panel keeping its own history.

kis-balazs, 2022-02-07:
I would like to be able to see previous points on the map, but I don't know whether can I or not influence this from Node Playground. If I could, is there a sample, example? If I cannot, would it be possible to solve it without touching the ROS bag?

kis-balazs, 2022-02-21:
Hello @defunctzombie @jtbandes!

Could we get some feedback on this, is there any reality in which this would happen/would be in plan, since it would be a **very** useful feature :)

jtbandes, 2022-10-25:
#3721 may help with the User Scripts (nee Node Playground) part of this, but allowing the panel to keep its own history is still a good request. I’ve marked #4703 a duplicate of this one.

romainreignier, 2022-12-12:
When using the sample data from Foxglove in the Map panel, we can see the whole trajectory:
[image]
But when using a rosbag, we can only the last point:
[image]
What is the difference of the 2 methods?
My bag lasts 130 sec while the demo data only 19 sec. Does it count because it preloads only few seconds?

romainreignier, 2022-12-12:
Fom the docs, it should work: https://foxglove.dev/docs/studio/app-concepts/playback#preloading
But note that I am using a MessageConverter from a custom ROS1 GPS message to `foxglove.LocationFix`.

defunctzombie, 2022-12-12:
@jtbandes I think we can close this feature as complete regarding user scripts and mcap/bag files. Since this issue was created we've added preloaded message processing to user scripts. This means that when you write a user script that creates map panel compatible messages and then open a file we will process and show all the processed messages.

@romainreignier What you describe sounds like it could be a new issue (possibly a bug) regarding the handing of message converters. Please open a new bug issue describing your workflow and what is and is not working. Please provide some sample data we can use to reproduce the issue.

I'm closing this issue as completed. For folks participating on this thread - please open new issues with specific items that are not working.

romainreignier, 2022-12-12:
Ok, done in https://github.com/foxglove/studio/issues/4966