foxglove/studio #2798: Add color to Node Playground Message.sensor_msgs__NavSatFix messages

kis-balazs, 2022-02-07:
Would there be a possibility to add color to a specific type of GPS message either by the type Message.sensor_msgs__NavSatFix or in any other way, when trying to use multiple GPS devices on the same map?

jtbandes, 2022-02-09:
This is a good feature request. I think it's not specific to Node Playground, but just a feature that we should add to the Map panel. Each topic can have a separate color, like each dataset does in the Plot panel.

kis-balazs, 2022-02-09:
Thanks, it seemed useful to me since I am using multiple GPS sensors at a time and it is very-very hard to follow which one is which.

When could we expect updates about this topic?

kis-balazs, 2022-02-21:
Hello @jtbandes!

Could we have a short update/feedback on this idea? :)

amacneil, 2022-02-21:
@defunctzombie thoughts on adding this to `foxglove.LocationFix` (potentially in addition to allowing topic color override, similar to marker messages)?

defunctzombie, 2022-02-21:
@amacneil The option I'd probably start with is what we do in the plot panel - which is to round-robin different colors when we see multiple topics. I'd probably leave this up to configure in the layout rather than the message.

defunctzombie, 2022-02-21:
@kis-balazs We've added this to the March product roadmap plan.

kis-balazs, 2022-02-21:
@defunctzombie sounds lovely, thank you very much! Looking forward to it! :)

amacneil, 2022-02-28:
@kis-balazs do you need the ability to customize the color per-message, or only a single color per-topic?

kis-balazs, 2022-02-28:
I would suggest color/topic would be a first approach. More relevant, from my pov. Thanks!

PR #2930 (2022-03-01): Map panel: use distinct color per topic
**User-Facing Changes**
The Map panel now displays each visible topic in a different color.

**Description**
Closes #2798

Test data: [multigps.mcap.zip](https://github.com/foxglove/studio/files/8158061/multigps.mcap.zip)

Colors for all points and current point:
[image]

Colors also shown in layer control:
[image]

Covariance and hovered point colors:
[image]
commit 58ed25fe (2022-03-01): Map panel: use distinct color per topic (#2930)

**User-Facing Changes**
The Map panel now displays each visible topic in a different color.

**Description**
Closes #2798
closed 2022-03-01 by jtbandes