foxglove/studio #2808: 3D pannel shows above full screen panel

clementchef, 2022-02-09:
3D panel sometimes pop on top of the full screen panel.
[image]

- Version: 0.25.1
- OS: ubuntu 18

**Steps To Reproduce**
Not completly sure. Seems pretty random. This was in live mode (not bag mode).

**Expected Behavior**

**Actual Behavior**

defunctzombie, 2022-02-10:
I've tried to reproduce this on my Mac in chrome but haven't been able to so far. I also tried with the desktop app on my Mac and could not reproduce (trying a few things like changing window size).

Any steps you can come up with to reproduce would be helpful. It's also possible its specific to running on ubuntu linux and has something to do with your video graphics.

Another thing I'd recommend is updating to a newer version of studio. We're on 0.29.0 now. I am not recalling anything specific we fixed with full screen panels but its possible there were some changes that have impacted this layering.

amacneil, 2022-02-28:
@clementchef closing per your Slack message - please reopen if this starts happening again.

closed 2022-02-28 by amacneil