foxglove/studio #2823: Display a 3D panel topic warning when header.stamp is very far in the future

jhurliman, 2022-02-10:
A common source of visualization failure is running a simulation where `/clock` is published with a timestamp starting at zero or some time in the past, but markers or other topics are published with a header.stamp set to the current local wall clock time. When a topic header.stamp is significantly in the future (significant is debatable here, I propose 30 seconds as a starting point) we should show a topic warning.

The distance into the future could also be user-configurable, and if the user set it to 0 it would catch anything published with a future timestamp. This can be helpful debugging in some cases, while it would break some normal use cases such as small amounts of clock skew and interpolation.

defunctzombie, 2022-02-10:
Related: https://github.com/foxglove/studio/issues/2575

jhurliman, 2022-07-14:
In New3D the transform library has been modified to allow unlimited forward/backward time deltas. Your visualization may be unexpected if there is a large clock disagreement, but it will still render.

closed 2022-07-14 by jhurliman