foxglove/studio #2824: Clear all warnings button

bertaveira, 2022-02-10:
Would be nice to have a way to clear all warnings or even ignore future ones. When things go sideways I have to restart foxglove since it is impossible to clear all. It is even hard to debug the situation since it is covering part of the screen.

[image]

defunctzombie, 2022-03-01:
@amacneil Let's add this to an upcoming roadmap month. I think its not a big lift for us to make this at least less terrible for now. Longer term I'd like to not use this style of notification for this issue - but the architecture we have today makes that trickier.

amacneil, 2022-03-01:
A few thoughts:
- "clear all" button would definitely help, although with these errors it looks like they will keep happening if the data source is producing them
- grouping errors by text might be nice, although you still have to have the annoying popup over your screen the whole time (assuming it just comes back after you clear it)
- nicest option is probably to move this warning to the data source sidebar (similarly to how we display data source errors)

defunctzombie, 2022-03-07:
We can tackle this by changing the MessageOrderTracker `update` call to inject problems into the player state and return a new player state. This would remove send notification and allow the issue to display on the data source sidebar in a less intrusive manner.

https://github.com/foxglove/studio/blob/main/packages%2Fstudio-base%2Fsrc%2Fcomponents%2FMessagePipeline%2FMessageOrderTracker.ts#L48

PR #3030 (2022-03-16): Remove sendNotification from MessageOrderTracker
**User-Facing Changes**
When encountering message order warnings the user is no longer bombarded with toast noticiations. Instead, the warnings are added to player problems and displayed alongside the existing player problems display (currently in the Data Source sidebar).

**Description**
Rather than emitting a notification for each message order violation, the MessageOrderTracker builds up an array of player problems. These player problems are added to the existing player problems. Downstream consumers of the playerState are responsible for th [... cut]
commit 5c0bb349 (2022-03-16): Remove sendNotification from MessageOrderTracker (#3030)

Rather than emitting a notification for each message order violation, the MessageOrderTracker builds up an array of player problems. These player problems are added to the existing player problems. Downstream consumers of the playerState are responsible for the display of the problems.

Fixes: #2824
closed 2022-03-16 by defunctzombie