foxglove/studio #2852: Data Platform data source name could be more helpful

amacneil, 2022-02-16:
**Description**
Currently when viewing remote Data Platform data, we give you a pretty strange display of the data source name.

Also, if the field is truncated we should show the full value on hover.

[image]

foxymiles, 2022-02-18:
What should we display here? Maybe just "Data Platform" or "Foxglove Data Platform"?

defunctzombie, 2022-02-18:
The `name` field in playerState is what gets put here. That field supports newline characters: `\n` for display (we use this with the nuscenes sample). I'd go with:

```
Foxglove Data Platform
<device name> (device id)
```