foxglove/studio #2853: Add more obvious topic settings for overriding marker color

amacneil, 2022-02-16:
**Description**
Currently, in the topic settings dialog, we don't visually show whether we are using an overridden color or inheriting the color from the marker. We should make this a checkbox or radio or something to more clearly differentiate between "use marker colors" and "override color".

[image]

foxymiles, 2022-07-25:
@amacneil This seems much improved in new3d. Maybe not worth fixing in old 3d?

amacneil, 2022-07-26:
It seems I cannot override the marker color at all in new3d.

[image]

I created https://github.com/foxglove/studio/issues/3943 to track the request. 

In the meantime, I will close this ticket because we are not doing any new development on old 3D.

closed 2022-07-26 by amacneil