foxglove/studio #2862: The Plot's timestamp (ROS2 Connection) is wrong

taiyuzhuo, 2022-02-17:
**Description**
It is shown in the below figure that the timestamp is not strictly increase monotonically。
[image]

- Version: 1.0.0
- OS: ubuntu 20.04

**Steps To Reproduce**

the talker node's code is here:
```python 
import rclpy
from rclpy.node import Node

from std_msgs.msg import String
from std_msgs.msg import Int16

class MinimalPublisher(Node):

    def __init__(self):
        super().__init__('minimal_publisher')
        self.publisher_ = self.create_publisher(Int16, 'topic', 10)
        timer_period = 0.1  # seconds
        self.timer = self.create_timer(timer_period, self.timer_callback)
        self.i = 0

    def timer_callback(self):
        msg = Int16()
        #msg.data = 'Hello World: %d' % self.i
        msg.data = self.i
        self.publisher_.publish(msg)
        self.get_logger().info('Publishing: "%s"' % msg.data)
        self.i += 1

def main(args=None):
    rclpy.init(args=args)

    minimal_publisher = MinimalPublisher()

    rclpy.spin(minimal_publisher)

    # Destroy the node explicitly
    # (optional - otherwise it will be done automatically
    # when the garbage collector destroys the node object)
    minimal_publisher.destroy_node()
    rclpy.shutdown()

if __name__ == '__main__':
    main()
```

**Expected Behavior**
when in the ros2 connection, the timestamp of the topic plot panel is strictly increase monotonically and synchronized with the timestamp of ros2 msg.

**Actual Behavior**
 the timestamp is not strictly increase monotonically.

jhurliman, 2022-02-17:
Root cause: https://github.com/foxglove/rtps/pull/16

closed 2022-02-17 by jhurliman

taiyuzhuo, 2022-02-23:
> Confirming I was able to reproduce with your example node. Thank you.
> 
> @jhurliman sending your way to take a look at this since its ros2 native connection related.

I have already update the foxglove studio to 1.1.0, but it seems like the previous problem is still occurs. 
[image]

defunctzombie, 2022-02-23:
@taiyuzhuo This fix is not yet in a release; it was merged after 1.1.0 was released. It will be in the next release tho!

taiyuzhuo, 2022-02-23:
> This commit [1b8508f](https://github.com/foxglove/studio/commit/1b8508f060f2f0b1aada1ccd908fe3d88fd6f4d5) was not yet included in the latest release, it will be in the next release. You can check by looking at the list of branches/tags under the commit message:
> 
> [image]
> 
> When it is included in a release it will look like this: [image]

Sorry.....