foxglove/studio #2868: Excessive memory usage when using plot on v1.0

airfield20, 2022-02-17:
**Description**
I have a view set up, streaming live data with rosbridge to foxglove studio v1.0. The view contains a console view and two plots. Each plot is subscribed to two topics that publish around 100 to 500hz. Over the course of about 30 mins my RAM utilisation will increase from around 150mb to around 1.3gb will streaming this data. Foxglove studio continually slows down over time and will eventually become unusable. Switching the view to another setup and then back to the plot fixes the issue temporarily.

- Version: 1.0 (desktop, not web)
- OS: Ubuntu 20.04

**Steps To Reproduce**

Create a plot view of a high frequency data value from a live sensor. Set the x axis range to 30 seconds. Let this run for a long time while watching a graph of your RAM usage. 

**Expected Behavior**
Old plot points will be deleted when they are no longer needed.

**Actual Behavior**
It seems old plot points must be kept mistakenly, or something other data being generated upon reading a message is taking up a ton of ram.

jtbandes, 2022-02-17:
Are you using rosbridge with ROS 1 or ROS 2? And are you running ROS on the same machine where you're running Studio?

airfield20, 2022-02-17:
I am using ROS 1. I do not have any nodes or a rosmaster running on the machine that is running studio. 

The robot is running ROS1 which has 1 computer running the master and a few other computers connected to the sensors. I am running rosbrige on the computer that is running the master.

defunctzombie, 2022-02-23:
@airfield20 Thanks for surfacing this! Indeed the "fix" I thought I did was not a real fix. I've done a correct fix in #2897 and confirmed the change for web and desktop. The fix will be in the next desktop release.

PR #2897 (2022-02-23): Fix unbounded memory growth in plot panel for new messages

**User-Facing Changes**
Plot panel releases old messages from memory after they leave the view window. The user should notice stable memory usage with the plot panel.

**Description**
https://github.com/foxglove/studio/pull/1707 attempted to fix the memory growth issue but did the filtering in the wrong place. The filtering needs to happen in the addMessages call so the reducer does not continue to hang on to the message events indefinitely.

Fixes: #2868
commit 19e83f05 (2022-02-23): Fix unbounded memory growth in plot panel for new messages (#2897)

https://github.com/foxglove/studio/pull/1707 attempted to fix the memory growth issue but did the filtering in the wrong place. The filtering needs to happen in the addMessages call so the reducer does not continue to hang on to the message events indefinitely.

Fixes: #2868
closed 2022-02-23 by defunctzombie