foxglove/studio #2875: Ability to publish standard ROS messages from Node Playground

amacneil, 2022-02-18:
1. Build and run [ros1-bench-workspace](https://github.com/foxglove/ros1-bench-workspace)
2. Connect in Studio via rosbridge
3. Open node playground and load the "Markers" template node.

Studio does not know about `visualization_msgs/Marker`:

[image]

We should ship these standard messages by default even if they are not currently being published by the ROS source.

defunctzombie, 2022-02-18:
This should also include the foxglove types: `foxglove.LocationFix`

jtbandes, 2022-02-18:
Isn't this a duplicate of https://github.com/foxglove/studio/issues/1261? Why was that one closed?

defunctzombie, 2022-02-18:
@jtbandes This is for the new syntax `Message<...>`. That ticket was for the old syntax and we did implement that feature for the new syntax.

defunctzombie, 2022-02-18:
Relevant lines of code: https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/UserNodePlayer/index.ts#L150

commit 735fff89 (2022-02-19): Add builtin types to node playground `Message` type (#2878)

**User-Facing Changes**
Using `Message<"visualization_msgs/Marker">` or `Message<"foxglove.LocationFix">` works even if your data source does not contain the datatype. This makes it possible to write nodes that emit these messages from messages of some other datatype.

**Description**

Fixes: #2875