foxglove/studio #2880: Teleop Panel / stop sign when hovering over button

robvoi, 2022-02-21:
I can't click the buttons on the teleop panel. Not sure what I might do wrong. Besides defining the topic there isn't anything to configure.
When I hover over the button, the mouse courser changes into a stop symbol.

- Version:
- OS: Ubuntu 20.03

**Steps To Reproduce**
- Add teleop panel
- configure topic
- buttons can't be pressed (stop sign)

robvoi, 2022-02-21:
I connected a ROS2 data source. Topics list and published messages get displayed.

defunctzombie, 2022-02-21:
Unfortunately our ROS2 connection does not support publishing yet. If you need publishing with ROS2, I'd recommend using the [rosbridge](https://github.com/RobotWebTools/rosbridge_suite) connection for now.

I've opened a feature request issue https://github.com/foxglove/studio/issues/2882 for adding publishing support.

Until we add publishing support to ros2 native connections, we should improve the UX around the teleop panel and errors so it is obvious why it does not work.

robvoi, 2022-02-21:
Thanks for the feedback. Falling back on the ros teleop packages then.
As the code exists in those packages, maybe they can be the base for an integration? (if allowed license wise)

PR #2892 (2022-02-23): Add better error states to teleop panel.
**User-Facing Changes**
This improves the presentation of the disabled state in the teleop panel.

**Description**
This checks if the user has connected to a publishable data source and has entered a topic and displays more useful error messages if not.

[image]
[image]

Fixes #2880