foxglove/studio #2883: Awkward floating icon

defunctzombie, 2022-02-22:
The inspect object icon feels awkwardly placed. The existing zoom icon is a different distance from the bottom and side than this other icon.

It's also a different distance from the side as the other text overlay at the bottom. We should find some consistency in spacing with these UI elements.

[image]

foxymiles, 2022-03-01:
What should we do here? Maybe consolidate the zoom button and the inspect button into a common toolbar like the 3d panel?

defunctzombie, 2022-03-01:
@2metres has some ideas around the image panel buttons and dropdowns. This should figure into those ideas.