foxglove/studio #2886: Special characters in MCAP topic names cause issues.

jameskuszmaul-brt, 2022-02-22:
**Description**

I was converting some data to an MCAP file to try out Foxglove; I found that while MCAP appears to permit arbitrary UTF-8 strings for topic names, if I use a topic name with spaces, hyphens, or periods (or presumably other types of special characters), adding a field on that topic to a plot will result in the field simply not showing (I am able to add to the legend, but no data for that line actually displays on the plot).

- Version: 1.1.0 (ran `docker run --rm -p "8080:8080" ghcr.io/foxglove/studio:latest` on 2/17).
- OS: Using browser (Chrome)

**Steps To Reproduce**
Create an MCAP log with a topic that contains at least one space. I was noticing this with things of the format `/topic/name subtopic.Type` (for sundry reasons, the source data I'm working with uses a slightly different definition of "topic" where you need both a name and a type to uniquely resolve topics). If I replace all the spaces/`.`'s with `/`'s, then things work fine.

I can provide an exact mcap file for repro if needed, but that'll require me generating some non-private data.

**Expected Behavior**
Ideally, arbitrary topic names are allowed. At a minimum, the error is more obvious.

**Actual Behavior**
No obvious failure; instead data simply doesn't display.

jameskuszmaul-brt, 2022-02-22:
Ended up needing to make thing simpler to reproduce anyways, so I pulled some https://github.com/frc971/971-Robot-Code to play with.

The two below MCAP files are identical except for the topic names that they use (files are ~13 KB).
[good_topic_name.mcap.gz](https://github.com/foxglove/studio/files/8113781/good_topic_name.mcap.gz)
[bad_topic_name.mcap.gz](https://github.com/foxglove/studio/files/8113782/bad_topic_name.mcap.gz)

jtbandes, 2022-02-22:
Thanks for the report! This is a limitation with certain panels in the app, namely those which use [message path syntax](https://foxglove.dev/docs/studio/app-concepts/message-path-syntax). I agree the error should be made more obvious. Since supporting arbitrary topic names seems like a nice feature for the app to have I'm going to convert this to a feature request.

jameskuszmaul-brt, 2022-02-22:
Yeah, I don't have a pressing need to have arbitrary characters in the topic name (although it would be nice), but the lack of a better error message really threw me for a loop (especially since I was creating the MCAP logs myself and so was simultaneously trying to validate that I wasn't messing anything up on my end).