foxglove/studio #2887: Can't handle optional nested submessages when parsing MCAP file

jameskuszmaul-brt, 2022-02-22:
**Description**

If I generate an MCAP log with a channel whose type has multiple layers of nested sub-messages (in this schema, `MessageType.fetchers[0].channel.schema.root_table.name` is a potentially valid reference), but don't specify one of the sub-fields, foxglove studio barfs complaining about an undefined reference.

In this, case we have the below logfile:
[foo.mcap.gz](https://github.com/foxglove/studio/files/8113728/foo.mcap.gz)

Which just contains a single channel/schema and a single message on that channel. The schema for the channel is:

```
{"$schema":"https://json-schema.org/draft/2020-12/schema","properties":{"fetchers":{"items":{"properties":{"channel":{"properties":{"destination_nodes":{"items":{"properties":{"name":{"type":"string"},"priority":{"type":"number"},"time_to_live":{"type":"number"},"timestamp_logger":{"type":"number"},"timestamp_logger_nodes":{"items":{"type":"string"},"type":"array"}},"type":"object"},"type":"array"},"frequency":{"type":"number"},"logger":{"type":"number"},"logger_nodes":{"items":{"type":"string"},"type":"array"},"max_size":{"type":"number"},"name":{"type":"string"},"num_readers":{"type":"number"},"num_senders":{"type":"number"},"num_watchers":{"type":"number"},"read_method":{"type":"number"},"schema":{"properties":{"enums":{"items":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"documentation":{"items":{"type":"string"},"type":"array"},"is_union":{"type":"boolean"},"name":{"type":"string"},"underlying_type":{"properties":{"base_type":{"type":"number"},"element":{"type":"number"},"fixed_length":{"type":"number"},"index":{"type":"number"}},"type":"object"},"values":{"items":{"properties":{"documentation":{"items":{"type":"string"},"type":"array"},"name":{"type":"string"},"object":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"bytesize":{"type":"number"},"documentation":{"items":{"type":"string"},"type":"array"},"fields":{"items":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"default_integer":{"type":"number"},"default_real":{"type":"number"},"deprecated":{"type":"boolean"},"documentation":{"items":{"type":"string"},"type":"array"},"id":{"type":"number"},"key":{"type":"boolean"},"name":{"type":"string"},"offset":{"type":"number"},"optional":{"type":"boolean"},"required":{"type":"boolean"},"type":{"properties":{"base_type":{"type":"number"},"element":{"type":"number"},"fixed_length":{"type":"number"},"index":{"type":"number"}},"type":"object"}},"type":"object"},"type":"array"},"is_struct":{"type":"boolean"},"minalign":{"type":"number"},"name":{"type":"string"}},"type":"object"},"union_type":{"properties":{"base_type":{"type":"number"},"element":{"type":"number"},"fixed_length":{"type":"number"},"index":{"type":"number"}},"type":"object"},"value":{"type":"number"}},"type":"object"},"type":"array"}},"type":"object"},"type":"array"},"file_ext":{"type":"string"},"file_ident":{"type":"string"},"objects":{"items":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"bytesize":{"type":"number"},"documentation":{"items":{"type":"string"},"type":"array"},"fields":{"items":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"default_integer":{"type":"number"},"default_real":{"type":"number"},"deprecated":{"type":"boolean"},"documentation":{"items":{"type":"string"},"type":"array"},"id":{"type":"number"},"key":{"type":"boolean"},"name":{"type":"string"},"offset":{"type":"number"},"optional":{"type":"boolean"},"required":{"type":"boolean"},"type":{"properties":{"base_type":{"type":"number"},"element":{"type":"number"},"fixed_length":{"type":"number"},"index":{"type":"number"}},"type":"object"}},"type":"object"},"type":"array"},"is_struct":{"type":"boolean"},"minalign":{"type":"number"},"name":{"type":"string"}},"type":"object"},"type":"array"},"root_table":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"bytesize":{"type":"number"},"documentation":{"items":{"type":"string"},"type":"array"},"fields":{"items":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"default_integer":{"type":"number"},"default_real":{"type":"number"},"deprecated":{"type":"boolean"},"documentation":{"items":{"type":"string"},"type":"array"},"id":{"type":"number"},"key":{"type":"boolean"},"name":{"type":"string"},"offset":{"type":"number"},"optional":{"type":"boolean"},"required":{"type":"boolean"},"type":{"properties":{"base_type":{"type":"number"},"element":{"type":"number"},"fixed_length":{"type":"number"},"index":{"type":"number"}},"type":"object"}},"type":"object"},"type":"array"},"is_struct":{"type":"boolean"},"minalign":{"type":"number"},"name":{"type":"string"}},"type":"object"},"services":{"items":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"calls":{"items":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"documentation":{"items":{"type":"string"},"type":"array"},"name":{"type":"string"},"request":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"bytesize":{"type":"number"},"documentation":{"items":{"type":"string"},"type":"array"},"fields":{"items":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"default_integer":{"type":"number"},"default_real":{"type":"number"},"deprecated":{"type":"boolean"},"documentation":{"items":{"type":"string"},"type":"array"},"id":{"type":"number"},"key":{"type":"boolean"},"name":{"type":"string"},"offset":{"type":"number"},"optional":{"type":"boolean"},"required":{"type":"boolean"},"type":{"properties":{"base_type":{"type":"number"},"element":{"type":"number"},"fixed_length":{"type":"number"},"index":{"type":"number"}},"type":"object"}},"type":"object"},"type":"array"},"is_struct":{"type":"boolean"},"minalign":{"type":"number"},"name":{"type":"string"}},"type":"object"},"response":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"bytesize":{"type":"number"},"documentation":{"items":{"type":"string"},"type":"array"},"fields":{"items":{"properties":{"attributes":{"items":{"properties":{"key":{"type":"string"},"value":{"type":"string"}},"type":"object"},"type":"array"},"default_integer":{"type":"number"},"default_real":{"type":"number"},"deprecated":{"type":"boolean"},"documentation":{"items":{"type":"string"},"type":"array"},"id":{"type":"number"},"key":{"type":"boolean"},"name":{"type":"string"},"offset":{"type":"number"},"optional":{"type":"boolean"},"required":{"type":"boolean"},"type":{"properties":{"base_type":{"type":"number"},"element":{"type":"number"},"fixed_length":{"type":"number"},"index":{"type":"number"}},"type":"object"}},"type":"object"},"type":"array"},"is_struct":{"type":"boolean"},"minalign":{"type":"number"},"name":{"type":"string"}},"type":"object"}},"type":"object"},"type":"array"},"documentation":{"items":{"type":"string"},"type":"array"},"name":{"type":"string"}},"type":"object"},"type":"array"}},"type":"object"},"source_node":{"type":"string"},"type":{"type":"string"}},"type":"object"},"channel_index":{"type":"number"},"count":{"type":"number"},"latency":{"properties":{"average":{"type":"number"},"max":{"type":"number"},"min":{"type":"number"},"standard_deviation":{"type":"number"}},"type":"object"}},"type":"object"},"type":"array"},"name":{"type":"string"},"phased_loops":{"items":{"properties":{"count":{"type":"number"},"handler_time":{"properties":{"average":{"type":"number"},"max":{"type":"number"},"min":{"type":"number"},"standard_deviation":{"type":"number"}},"type":"object"},"name":{"type":"string"},"wakeup_latency":{"properties":{"average":{"type":"number"},"max":{"type":"number"},"min":{"type":"number"},"standard_deviation":{"type":" [... cut ...] s to work fine--I assume that https://github.com/foxglove/studio/blob/a20cf7ef51cae38d143a2d76e3e048c84c04a0d7/packages/mcap-support/src/parseJsonSchema.ts is somehow iterating over all the sub-types in a message when attempting to read things).

Also, empty JSON messages (`{}`) don't seem to be allowed, even if the schema permits it.

**Expected Behavior**

The logfile is successfully read and I can view the data that is populated.

**Actual Behavior**

I get an error and can't view any data :(

jtbandes, 2022-02-23:
Here's a PR to address the issue: https://github.com/foxglove/studio/pull/2898

Other parts of the app were not originally built to handle optional fields, so you might still run into other issues elsewhere. Please feel free to file more tickets as you encounter problems!

By the way it's pretty cool that you're using MCAP and Foxglove Studio with a FRC robot! 😄  A couple of us at Foxglove are FRC alumni / former mentors. I've often thought it would make sense to have some kind of NetworkTables integration with Studio, but recording MCAP on the robot is a great idea too. Is there somewhere we can learn more about what you're working on?

closed 2022-02-23 by jtbandes

jameskuszmaul-brt, 2022-02-23:
Hah, I'm actually investigating this as part of a commercial application for Blue River Technology (Adam Cadien suggested it to me). But the middleware layer that we use is the same one that those of us on 971 developed (see https://github.com/frc971/971-Robot-Code). That middleware ("AOS") exists specifically because ROS and other similar things don't quite have the design to support the needs we've tended to have.

AOS uses flatbuffers for serialization and has its own logfile format. I'm just converting to MCAP because it's easier (for now) to post-process the logs and generate MCAP than it is to write the typescript to parse an AOS log directly.
I've been trying to improve the visualization tooling we have attached to AOS for a bit--for the FRC team, I made a simple time-series plotter based on WebGL (which does actually directly read flatbuffers from typescript, but that requried me to write a [reflection library](https://github.com/frc971/971-Robot-Code/blob/master/aos/network/www/reflection.ts) for that...). But I'd like to not be unnecessarily trying to maintain/build visualization tooling that may already exist, and so took Adam's suggestion to take a look at this and see how well it integrates.

jameskuszmaul-brt, 2022-02-23:
If it does look like an AOS<->foxglove integration works well, I will probably try to get everything relevant open-sourced so that we have a clean visualization answer for both the FRC team and anyone else that may end up wanting to use AOS.

jameskuszmaul-brt, 2022-02-23:
Oh and an intro to AOS, which should still be largely correct, if you care: https://www.youtube.com/watch?v=DqnjbtQBS3w

jtbandes, 2022-02-23:
Awesome, thanks for all the references. Feel free to also provide any feedback on the https://github.com/foxglove/mcap repo or in our [Slack group](https://foxglove.dev/slack) if you run into issues with the readers/writers or the spec. It's still pretty new and we're happy to have others trying it out!

jtbandes, 2022-02-23:
By the way, not sure if we've explained this in our docs anywhere, but https://studio.foxglove.dev is kept up to date from our main branch, so you can see the changes live starting a few minutes after the PR has merged.

jameskuszmaul-brt, 2022-02-24:
Wrote up the suggestion as #2909, so that I have it documented.