foxglove/studio #2896: Memory growth when dropping bags onto studio.

foxymiles, 2022-02-23:
**Description**
Repeatedly dropping bag files onto studio seems to result in continuous memory growth that suggests a memory leak.

- Version:  1.0.0
- OS: macOS

**Steps To Reproduce**

1. Open studio.foxglove.dev
2. Select "explore new data".
3. Open the chrome memory debugger to check memory use.
4. Repeatedly drag nuscenes.bag onto studio

**Expected Behavior**
Observe that memory continues to grow each time the bag is dropped on studio.

**Actual Behavior**
Memory should be reclaimed when loading a new bag.

amacneil, 2022-02-28:
@defunctzombie do you feel after #2904 this is sufficiently addressed, or we should track down other sources of leaks?

defunctzombie, 2022-02-28:
@amacneil I think that #2904 sufficiently addresses this issue. I'm closing the issue. @jtbandes if you think there's more here lets make a new issue that isn't a bug for any additional tweaks.

closed 2022-02-28 by defunctzombie