foxglove/studio #2901: Message count and frequency are missing for Data Platform sources

amacneil, 2022-02-23:
**Description**
[image]

Version: web

These are currently missing for data platform data. If they aren't available, we should remove these columns.

foxymiles, 2022-02-28:
@amacneil I can count the messages per topic pretty easily but they'll show as 0 for topics we haven't subscribed to. Is that ok?

[image]

jtbandes, 2022-02-28:
We should show "unknown" or "–" or something if we don't have the information, rather than 0.

PR #2923 (2022-02-28): Show dashes for missing message counts for data platform topics.
**User-Facing Changes**
This displays dashes if message counts are not available for a data source.

Fixes #2901
closed 2022-02-28 by foxymiles