foxglove/studio #2905: Could you please provide a more detailed extension example？

rogue-shadowdancer, 2022-02-24:
I am trying to write my first foxglove extension( In order to better distinguish multiple 3D graphics, I am going to add a dark outline to each cube represented by visualization_msgs/MarkerArray, for example:
[image]
actually they are made of many small cubes ),
but unluckily I did not know anything about front end（such as javascript/typescript/react/...). And they are also not used in my daily works(usually we use C++ & python), which means even I spent a few days learning that, I still do not know how these things works(sorry for my poor coding skill..). 
So in order to let a beginner start their job in a few hours, and finish them in a few days(Because some of the functions themselves did not looks difficult), could you please provide a react version of turtle panel extension, and explained apis in a more detailed way? Because all of the api you introduced in docs look like under react in typescript. 
Or do you have some other good ideas about this?

foxymiles, 2022-07-25:
I've added a few more example panels in https://github.com/foxglove/create-foxglove-extension/tree/main/examples. Are these helpful?

rogue-shadowdancer, 2022-08-17:
Great! That helps a lot!
Thanks!

foxymiles, 2022-08-17:
You're welcome! Closing this for now. If there are specific things that are unclear in the examples or the API please let me know on slack.

closed 2022-08-17 by foxymiles