foxglove/studio #2906: ROS bag playback - Warning: "Very large block found"

yuanli19, 2022-02-24:
When replaying a ros bag, the following warning appears: 

```
Very large block found

A very large block (1202ms) was found: 112MB. Too much data can cause performance problems and even crashes. Please fix this where the data is being generated.
```

- Version: 1.0.0
- OS: Ubuntu 20.04

 
Though this seems to be related to the data source, the warning message in foxglove studio does not help me narrow down the problem. What does "very large block" mean? The playback in rviz works fine. With foxglove, the playback gets stuck frequently for the same bags.

Splitting up the rosbags did not change the behavior. 

Suggestion:
Add verbose information about what is causing this, and how it might be fixed would be helpful in the warning message.

rogue-shadowdancer, 2022-02-24:
The same problem also occurred to me.  Seems that some large msgs exists in this bag(large pictures or something else), cause they are frequently appear in those bags have 1080p pictures or grid maps.
But I still wander how to deal with this problem.

amacneil, 2022-02-28:
Is it possible for either of you to share a bag file demonstrating this (either on Github, or privately via our [Slack channel](https://foxglove.dev/slack))? 

cc @esthersweon we may need to better document this

rogue-shadowdancer, 2022-03-01:
@lanyusea could please ask Power to find one here? I think some bag with fisheye pictures or grid maps may be suitable.

lanyusea, 2022-03-03:
@rogue-shadowdancer found a fish-eye bag but didn't reproduce the warning message, will keep an eye on that.

but the replaying (from local disk) is surely lag and lagger than rviz. I do think it may be related to the file I/O method. @amacneil  will sent the bag to you later by email

defunctzombie, 2022-03-03:
**tl;dr** I think we should remove this error.

I am working on a new bag player and have looked into where this message comes from. When we load bag files into studio, we also cache the messages into memory across the entire bag (well, as much as we can reasonably use the available memory). The caching logic uses a concept of "blocks" to store messages in chunks and tracks the memory use by each block. These "blocks" are also what power viewing plot data or gps data across the entire bag. There is nothing inherently "slow" with having a large block since that could just be the result of having large messages (i.e. images). Such large blocks will ultimately impact how much we can cache in memory which limits how much we can display on a plot chart for example. This also ties into an open feature we need to work on which is for panels to indicate whether they need their topics "preloaded" in this manner. Panels like the image panel do not need images pre-loaded across the entire dataset range while panels like plot do. Allow the panel to specify this would let our caching layer avoid loading all the large images which would also avoid any "slowdown".

amacneil, 2022-03-03:
@lanyusea thanks I received the file

amacneil, 2022-03-10:
@defunctzombie will this be fixed by https://github.com/foxglove/studio/pull/2915?

defunctzombie, 2022-03-10:
@amacneil yes - the message is removed in the new experimental player as the byte size of a block does not impact the performance of storing the block.

defunctzombie, 2022-03-11:
For anyone experiencing this issue - please give the new `Experimental Bag Player` a try. It is not yet in a desktop release but available on web.

The setting is in the app settings.

[image]

commit 3138944e (2022-04-25): Enable Experimental bag player as the default (#3249)

The experimental bag player is ready to graduate to the default. With this player users get better
seek behavior. When a user seeks, the player will backfill the last message on every topic. This
results in a better experience for panels which rely on having the last message on topics to display
useful info (which is basically every panel).

The previous bag player would backfill messages within a certain small window prior to the seek position.
If a message was outside that window it would not get backfilled. This meant some topics with i [... cut]
closed 2022-04-25 by defunctzombie