foxglove/studio #2916: Node playground type checking failing on start when loading from data platform

ccpowers, 2022-02-25:
**Description**
 When loading a new bag from the data platform, node playground incorrectly reports errors on types for input topics. Below is an example using node playground to transform `geometry_msgs/PoseArray` messages on topic `/particlecloud` into `visualization_msgs/MarkerArray` messages on topic `/studio_node/particle_cloud`. 
 When loading a bag from disk with these messages, the node playground reports no errors. If the same bag is uploaded into data platform and opened in studio via data platform, the errors shown below are reported. If the node playground is edited and re-saved (adding trivial changes to comments, or other irrelevant changes), the errors disappear. 
 
[image]

- Version: In browser (chrome)
- OS: The browser

**Steps To Reproduce**
1. Add a node playground panel with the following node included:
```
// The ./types module provides helper types for your Input events and messages.
import { Input, Message } from "./types";

// Your  node can output well-known message types, any of your custom message types, or
// complete custom message types.
//
// Use `Message` to access your data source types or well-known types:
// type Twist = Message<"geometry_msgs/Twist">;
//
// Conventionally, it's common to make a _type alias_ for your node's output type
// and use that type name as the return type for your node function.
// Here we've called the type `Output` but you can pick any type name.
type MarkerArray = Message<"visualization_msgs/MarkerArray">;
type Marker = Message<"visualization_msgs/Marker">;
type Pose = Message<"geometry_msgs/Pose">;
type PoseArray = Message<"geometry_msgs/PoseArray">;
type Header = Message<"std_msgs/Header">;

// These are the topics your node "subscribes" to. Studio will invoke your node function
// when any message is received on one of these topics.
export const inputs = ["/particlecloud"];

// Any output your node produces is "published" to this topic. Published messages are only visible within Studio, not to your original data source.
export const output = "/studio_node/particle_cloud";

// This function will transform one pose into one marker
const scale: Message<"geometry_msgs/Vector3"> = {
  x: 0.01,
  y: 0.01,
  z: 0.01,
};

const lifetime = { sec: 0.02, nsec: 0 };

const color: Message<"std_msgs/ColorRGBA"> = {
  r: 100,
  g: 0,
  b: 0,
  a: 100,
};
function poseToArrowMarker(pose: Pose, header: Header, id: number): Marker {
  return {
    header: header,
    ns: "particlecloud",
    id: id,
    type: 0, // arrow
    action: 0, // add
    pose: pose,
    scale: scale,
    color: color,
    lifetime: lifetime,
    frame_locked: true,
    points: [],
    colors: [],
    text: "",
    mesh_resource: "",
    mesh_use_embedded_materials: false,
  };
}
// This function is called with messages from your input topics.
// The first argument is an event with the topic, receive time, and message.
// Use the `Input<...>` helper to get the correct event type for your input topic messages.
export default function node(event: Input<"/particlecloud">): MarkerArray {
  const markers: MarkerArray = {
    markers: [],
  };

  for (let i = 0; i < event.message.poses.length; i++) {
    markers.markers.push(
      poseToArrowMarker(event.message.poses[i], event.message.header, i)
    );
  }
  return markers;
}
```

2. Upload a bag with messages on `/particlecloud` of type `geometry_msgs/PoseArray` to data platform (I will attach an example in the comments in a moment)
3. Open the bag in studio via the data platform ui 

**Expected Behavior**
Opening any bag with messages on `/particlecloud` from disk or from data studio will pass type checking

**Actual Behavior**
Opening a bag from disk passes type checking, but opening the same bag from data platform does not

ccpowers, 2022-02-25:
This contains an example bag, in a `.zip` because GitHub won't upload `.bag` files. 
[Uploading particlecloudexample.zip…]()
.

ccpowers, 2022-02-25:
In addition, i've tried to skirt this issue by writing my own type to replace the `Input<"/particlecloud">` type as shown below:
```
// Re-create Input type with message def
type PoseArrayEvent<T> = {
  topic: T;
  receiveTime: Time;
  message: Message<"geometry_msgs/PoseArray">;
};

// This function is called with messages from your input topics.
// The first argument is an event with the topic, receive time, and message.
// Use the `Input<...>` helper to get the correct event type for your input topic messages.
export default function node(
  event: PoseArrayEvent<"/particlecloud">
): MarkerArray {
  const markers: MarkerArray = {
    markers: [],
  };
  //
  for (let i = 0; i < event.message.poses.length; i++) {
    markers.markers.push(
      poseToArrowMarker(event.message.poses[i], event.message.header, i)
    );
  }
  return markers;
}
```

However, this results in the following error displaying instead, which also does not go away until I re-save the node playground node, even though I am playing the bag and the topic is available.
[image]

defunctzombie, 2022-02-25:
I've also observed this behavior on new connections. I think the key to fixing it is in your comment that any trivial change and re-save makes the error go away. I suspect its an ordering or re-compilation issue around selecting the data source.

commit 1e2cdcad (2022-03-03): Reset node playground workers when topics change.

Node playground nodes generate code for messages and inputs from the set of available topics and datatypes. This code generation would run once on startup. If the generation happened to run before the data source provided the topics and datatypes, the user would see compilation errors in their nodes and the nodes would not run.

This change re-generates the nodes when the data source topics change. This allows the node code to compile and the nodes to start running as the data source loads available topics and datatypes.

Fixes: #2916
commit fb83bcec (2022-03-03): Reset node playground workers when topics change.

Node playground nodes generate code for messages and inputs from the set of available topics and datatypes. This code generation would run once on startup. If the generation happened to run before the data source provided the topics and datatypes, the user would see compilation errors in their nodes and the nodes would not run.

This change re-generates the nodes when the data source topics change. This allows the node code to compile and the nodes to start running as the data source loads available topics and datatypes.

Fixes: #2916
PR #2939 (2022-03-03): Reset node playground workers when topics change.

**User-Facing Changes**
Node playground nodes are updated with generated types as the datasource loads new topics and datatypes.

**Description**
Node playground nodes generate code for messages and inputs from the set of available topics and datatypes. This code generation would run once on startup. If the generation happened to run before the data source provided the topics and datatypes, the user would see compilation errors in their nodes and the nodes would not run.

This change re-generates the nodes when the data source topics change. T [... cut]
closed 2022-03-03 by defunctzombie