foxglove/studio #2924: Cannot reconnect to Rosbridge after connection failure

rgov, 2022-02-28:
Foxglove Studio 1.1.0 on macOS 12.2.1

1. Run the Rosbridge server (ROS 1) and connect from Studio.
2. Kill the Rosbridge server. Studio notices this and displays an error.
3. Try to reconnect to the same data source from Studio. Studio doesn't appear to be able to reconnect. (Actually it does if you wait several minutes, probably for some timeout?)
4. Refresh Studio and try to connect again. It succeeds.

It would be especially nice if Studio just keeps trying to reconnect automatically, or makes it one click to reconnect. I'm re-running roslaunch all the time on my system and so the connection goes down frequently.

defunctzombie, 2022-02-28:
https://github.com/foxglove/studio/pull/2890 Will be in the next release and should resolve this issue. Studio would re-connect but fail to re-subscribe to topics.

defunctzombie, 2022-02-28:
v1.2.0 is out which includes the above fix. If that doesn't resolve the issue described please re-open this ticket.

rgov, 2022-04-04:
(Please re-open for me, I do not have privileges.)

I am still (now on 1.5.1) seeing suboptimal behavior here.

1. Run the Rosbridge server (ROS 1) and connect from Studio.
2. Kill the Rosbridge server. Studio notices this and displays an error.
3. Try to reconnect to the same data source from Studio. Rosbridge logging shows no client connecting. No messages appear in Studio UI.

rgov, 2022-04-04:
Appears there might be a 90 second timeout before the connection starts working.

defunctzombie, 2022-04-04:
> 3\. Try to reconnect to the same data source from Studio.

@rgov Is this explicitly done via the "open connection" dialog in Studio or by keeping the existing data source in Studio and waiting for it to reconnect?

Are you connecting to localhost or a separate computer?

rgov, 2022-04-04:
> Is this explicitly done via the "open connection" dialog in Studio or by keeping the existing data source in Studio and waiting for it to reconnect?

I use the dialog to connect again. I would *prefer* that it reconnect immediately.

> Are you connecting to localhost or a separate computer?

Technically it is localhost, but it is a port forwarded via a Visual Studio Code remote session (i.e., SSH port forwarding).

defunctzombie, 2022-04-06:
For me Studio re-connects immediately once rosbridge is launched. Here is a screen recording of me killing and re-starting rosbridge. Each time the connection is re-established immediately.

https://user-images.githubusercontent.com/84792/162080861-09ee3c7f-f4ac-497c-8dd2-531767a78e6d.mov

This is with Studio v1.6.0 which I think has the same logic for connectivity as v1.5.1

defunctzombie, 2022-04-12:
@rgov I was not able to reproduce (see above video). If you can describe what I should try differently happy to give it a try and figure out what behavior you are seeing or wanting to happen.

defunctzombie, 2022-04-19:
Closing this out since I've not heard back and was not able to reproduce. Happy to re-visit if there are steps I can use to reproduce.

closed 2022-04-19 by defunctzombie