foxglove/studio #2926: Plot jitters due to format of x-axis label

rgov, 2022-02-28:
Inherited from Webviz but still present on Foxglove Studio 1.1.0 on macOS 12.2.1

From this issue: https://github.com/cruise-automation/webviz/issues/544

---

[image]

This is a plot of a value over time. The plot keeps jumping because the rightmost x-axis label sometimes switches from 1 digit of precision to 2.

jhurliman, 2022-04-26:
Still present:

https://user-images.githubusercontent.com/195374/165198320-21ff1973-86f9-4be7-99f4-61414b64e927.mp4

defunctzombie, 2022-05-05:
@jtbandes assigning to you. I think your PR is 👍 and we can go from there.

closed 2022-05-05 by jtbandes