foxglove/studio #2931: Error boundary is not scrollable

jtbandes, 2022-03-01:
When an error appears, if the height of the error boundary is too short, the error is not visible, but it's impossible to scroll down and see the error:

[image]

When you increase the height, the error becomes visible:

[image]

jtbandes, 2022-08-24:
Reopening. If I have a small crashed panel, I can't access the dismiss/remove buttons:

[image]

[image]

foxymiles, 2022-08-30:
This is a bit tricky since the details themselves are also scrollable. Maybe we should make the entire container scrollable and just completely expand the details when shown?

PR #2940 (2022-03-02): Fix ErrorDisplay scrolling and styling tweaks
**User-Facing Changes**
Not worth calling out in release notes

**Description**
- Fix #2931 by making error boundary scrollable
- Use monospace font for error message
- Add a background to the error stack
- Fix the `cursor` style on links
- Remove some extraneous styles & elements
commit 0392a0e1 (2022-03-02): Fix ErrorDisplay scrolling and styling tweaks (#2940)

**User-Facing Changes**
Not worth calling out in release notes

**Description**
- Fix #2931 by making error boundary scrollable
- Use monospace font for error message
- Add a background to the error stack
- Fix the `cursor` style on links
- Remove some extraneous styles & elements
PR #4326 (2022-08-31): make error boundary container scrollable
**User-Facing Changes**
Not worth mentioning in release notes

**Description**
Fixes #2931

https://user-images.githubusercontent.com/14237/187754797-5efafa32-5ac5-4758-9ec6-abec951647f5.mov
commit 6b2db6ba (2022-08-31): make error boundary container scrollable (#4326)

**User-Facing Changes**
Not worth mentioning in release notes

**Description**
Fixes #2931
closed 2022-08-31 by jtbandes