foxglove/studio #2934: Image msg is not able to display on Image Panel

youngzhichun, 2022-03-01:
**Description**
Start from version 1.1, Image panel was not able to display videos. Retried with V1.2, issue still exists. However V1.0 looks good.

- Version:
- OS: Ubuntu16.04

**Steps To Reproduce**

Video msg def: contains ROS std Timestamp and self-defined timestamp(with sec and nsec as the only two variables.)  With uint8 array, encode as jepg format.

**Expected Behavior**

**Actual Behavior**

jtbandes, 2022-03-01:
Hi, can you share an example bag/mcap file?

jtbandes, 2022-03-01:
There were some changes to the image panel between 1.0 and 1.1:

- https://github.com/foxglove/studio/pull/2841
- https://github.com/foxglove/studio/pull/2851

And some changes between 1.1 and 1.2:
- https://github.com/foxglove/studio/pull/2861
- https://github.com/foxglove/studio/pull/2856

youngzhichun, 2022-03-03:
@jtbandes Hi, thank you so much for the quick response, and sharing those information! 

Unfortunately due to data privacy, I cannot share a bag with you. But I can provide more details instead.

# How we get the Image msg written in the rosbag:

1. Raw video is in h264 format, and frameIdx: timestamp was originally recorded in a txt
2. We took the video, compress it with c++ opencv lib. 
    - loop through the h264 video
    - cv::VideoCapture -> read() to get a frame, then compress it with cv::resize()
    - cv::imencode(".jpeg", ...) to write the video uint8 Array into rosbag
3. rosbag msg definition looks very similar the same with what you mentioned in [here](https://github.com/foxglove/studio/pull/2841), unfortunately I cannot share the exact definition with you. But we also have `timestamp`, `data`, `format` field, which I believe are essential to the video display on Foxglove Image Panel. However, we don't have `width` and `height` field.

# Why I assume there is something wrong with V1.1 and V1.2

The issue disappear with V1.0, and we also tried with Webviz, it looks good too. 

The issue only appears start from V1.1. 
1. With docker version: image panel seems not receiving any msg at all. Showing "Waiting for images on: /some/camera/topic/name"
2. With Ubuntun deb version: image panel seems to receive the msg (as the timestamp is populating), but the image panel is pure black.

I hope those information can provide more details. Thank you again!

youngzhichun, 2022-03-08:
Hi @defunctzombie 

Thanks for your information, data type is uint8 array, encoded as jpeg. We used cv::imencode to get the compressed image, and all parameters mentioned [here](https://docs.opencv.org/3.4/d8/d6a/group__imgcodecs__flags.html#ga292d81be8d76901bff7988d18d2b42ac) are set to default, except that we set cv.IMWRITE_JPEG_QUALITY to 60.

I tried to produce some dummy data. Unfortunately all raw image are directly taken from cameras on vehicles(h264 videos), there is really no way to get some dummy data without involving a whole lot more guys in this. 

Actually I am pretty satisfied with V1.0 available. If there's no one else complaining about a similar issue, let's close it for now.

closed 2022-03-08 by youngzhichun

jtbandes, 2022-03-08:
What is the _name_ of your custom image datatype? Only these datatypes are supported by the Image panel: https://github.com/foxglove/studio/blob/7f9398802f30bb6277b9ed7a9718508dbe777feb/packages/studio-base/src/panels/ImageView/lib/normalizeMessage.ts#L16-L23  I would be surprised if you were previously able to display a different (custom) datatype that was not in this list.

youngzhichun, 2022-03-09:
@jtbandes Hi, thanks for the clarification.
 Actually our custom image datatype is **NOT** on your list. It only contains essential fields like
- data: uint8array
- format: string
- header: ROS std header
- timestamps: with int32 sec and int32 nsec as its attribute
- etc.
However, it is able to display properly on V1.0.

jtbandes, 2022-03-09:
Even in v1.0, Studio would filter out image topics by datatype: https://github.com/foxglove/studio/blob/v1.0.0/packages%2Fstudio-base%2Fsrc%2Fpanels%2FImageView%2Findex.tsx#L285 

So this would filter down the list of topics available to select:
[image]

I know you said it displays properly on v1.0, but I don't understand how you could select your topic, because the topic shouldn't be available in the list. Did you edit the layout json manually, or did you previously have a layout that worked with a standard datatype and later change it to a custom datatype?

Anyway, you could try using [Node Playground](https://foxglove.dev/docs/studio/panels/node-playground) to translate the type to one that Studio understands. For example, translate the custom type to `sensor_msgs/CompressedImage`:

```ts
import { Input, Message } from "./types";

export const inputs = ["/camera"];
export const output = "/studio_node/camera_converted";

export default function node(event: Input<"/camera">): Message<"sensor_msgs/CompressedImage"> {
  return {
    header: event.message.header,
    data: event.message.data, 
    format: event.message.format,
  };
};
```

youngzhichun, 2022-03-10:
@jtbandes Thank you so much for all those information. I will tried it out and see if it works