foxglove/studio #2950: Studio publisher only connects to one subscriber (ros1 direct connection)

jessicaaustin, 2022-03-01:
**Description**

When publishing a message via Studio then the publisher only sends the message to one subscriber at a time.

This happens both via the Extensions API and the Publish panel.

Not sure if this bug should be posted here or at https://github.com/foxglove/ros1 but I'm happy to copy or move it if needed.

Related: https://github.com/foxglove/studio/issues/2057

**Steps To Reproduce**

- Version: Studio 1.1.0
- OS: Ubuntu (but i don't think this matters)
- ROS 1, live connection, all on one host

1. Run `rostopic echo /chatter` in two or more terminals, so you have at least two subscribers
2. Open Studio and add a Publish panel, and configure it to publish to `/chatter` with message type `std_msgs/String`
3. So at this point you should have one publisher and two or more subscribers on `/chatter`
4. Send a message via Studio

**Expected Behavior**

The message is received by all subscribers.

**Actual Behavior**

The message is only received by one subscriber.

There is also a warning in the dev console in Studio:

```
ROS1] failed to write 0 bytes to tcpros://127.0.0.1:35472: DataCloneError: Failed to execute 'postMessage' on 'MessagePort': ArrayBuffer at index 0 is already detached.
```

this error message shows up N-1 times, where N is the number of subscribers.

NOTE: I've also seen this same behavior when using the Extensions API to publish. And it doesn't matter if you use `rostopic echo` or a full-fledged node. 

[image]

SkyRoboticsLLC, 2022-03-02:
Having the exact same problem. Thanks for raising this up.

foxymiles, 2022-03-02:
Thanks very much for writing such a detailed bug report. It helps a lot. 

I'm able to reproduce this here so I'll take a look.

PR #2968 (2022-03-04): Use latest version of @foxglove/electron-socket to fix publishing issue
**User-Facing Changes**
This uses the latest version of @foxglove/electron-socket to fix an issue with publishing to multiple subscribers.

**Description**
This just upgrades to take advantage of a fix in the electron-socket library.

Fixes #2950
closed 2022-03-04 by foxymiles