foxglove/studio #2953: Cant zoom on plots while using live connection

clementchef, 2022-03-02:
Cant zoom on plots while using live connection

[image]

- Version: 1.1.0
- OS: Ubuntu 18.04

**Steps To Reproduce**
Plot something while in live connection mode. 
Try to scroll, pan or zoom on graph always revert to default view.
Sync option in panel setting is off.

**Expected Behavior**

**Actual Behavior**

jtbandes, 2022-03-02:
Seems related to #2701

foxymiles, 2022-03-02:
@clementchef Can you please share more detail on your datasource? I'm not seeing this with a simple rosbridge data source.

defunctzombie, 2022-03-10:
Via @clementchef - this is fixed in v1.3.0

closed 2022-03-10 by defunctzombie