foxglove/studio #2966: 2D/3D plot for trajectories

HugoFeng, 2022-03-03:
ref: https://facontidavide.github.io/PlotJuggler/visualization_howto/index.html#display-a-xy-curve
Can we have the plot panel display not only "1D" data, but also 2D or 3D trajectories.
For example in Plotjuggler, if one drag "pos.x, pos.y" into the plot, the plot will use them as axis, while the playback will update the "dot" showing the position at the moment.
This is very usefull and a must to have for working on Odometry for robotics. It'll be cooler if it can be extended to x,y,z while Plotjuggler only support x,y display.

jtbandes, 2022-03-07:
You might be able to accomplish what you are describing using the custom x-axis settings described here: https://foxglove.dev/docs/studio/panels/plot

[image]

HugoFeng, 2022-03-08:
> You might be able to accomplish what you are describing using the custom x-axis settings described here: https://foxglove.dev/docs/studio/panels/plot
> 
> [image]

Thank you, it works. glad to know this! Though I have 2 more questions:
1. how can I show a marker on my current position
2. It only shows the trajectory in within the "pre-fetched" time slot. Is there anyway it can pre-load the entire "accum" trajectory?
Thanks.