foxglove/studio #2974: Add dataType to MessageEvent in @foxglove/studio

jhurliman, 2022-03-04:
Incoming `MessageEvent`s (https://github.com/foxglove/studio/blob/1bfcfa924c7038050fcceb714d477921eb2e50d9/packages/studio/index.d.ts#L39) would be easier to dispatch to different handlers if they included a `dataType` field in addition to `topic`. Otherwise, I have to manage my own topic->dataType mapping.

defunctzombie, 2022-03-04:
Yea - I agree - I got annoyed at this when I did the message normalization in Image and Log panel recently.

Kind of a duplicate of https://github.com/foxglove/studio/issues/1923

The way I handled it for now in the panels I needed to tie them together was to use the topics array of the `Topic` type and store that with the extension or panel. Then I used that to lookup the datatype. So it is possible to do today - but was more annoying that it needed to be.

jtbandes, 2022-12-21:
Done in https://github.com/foxglove/studio/pull/4554

closed 2022-12-21 by jtbandes