foxglove/studio #2978: Showing Historical Data

valschmidt, 2022-03-06:
We like the feature that allows us to fix the time scale to the last X seconds. But we would like the size of the buffer of data available to display to be a separate value from the size of the plot, such that one could optionally interactively increase the time scale and look and pan back in time. 

For example, we envision specifying a time buffer of several hours and by default displaying the last 120 seconds. But when an operator desires, he/she can unlock the display and zoom out in time to see the last X hours of data or zoom back in to some small duration in that past record. RQT has the is ability, we’ve come to depend on it. 

As an example, consider engine coolant temperature or even fuel tank level. It can be hard to discern trends in these values over a short period of time, and it is useful to occasionally view them over the last several hours.

jtbandes, 2022-03-07:
To clarify, is this mostly about the Plot panel?

This could be a small adjustment to the settings provided by the Plot panel, or it could be satisfied by a larger project: https://github.com/foxglove/studio/issues/2769

valschmidt, 2022-03-07:
The plot panel, yes. 
I'm not sure of the pros/cons of doing this as part of pause/rewind functionality in #2769. It seems to me, it is not quite the same, in the sense that one needs to change the horizontal scale of the plot to see a longer time history all at once, rather than just rewind it.

jtbandes, 2022-03-07:
Yes, I suppose even if we implement #2769, we would still need an x range setting on the plot panel to easily control which part of the range is shown, even if the amount of available data is controlled by the app-level "DVR" setting.