foxglove/studio #2997: Message path autocomplete suggestions are not useful in some cases

foxymiles, 2022-03-09:
**Description**
The message path autcomplete suggestions in this file are not useful and lead to empty plots

- Version: 1.2.0
- OS: macOS

**Steps To Reproduce**
1. Load this mcap file into studio
2. Create a plot panel
3. Select one of the suggested plot paths like `/base_footprint.transforms[:]{child_frame_id==0}.transform.rotation.y`

[test.mcap.zip](https://github.com/foxglove/studio/files/8216627/test.mcap.zip)

**Current Experience**
The suggested paths result in empty plots.

**Desired Experience**
In this case studio should suggest something that will actually result in a useful plot like:

`/base_footprint.transforms[:]{child_frame_id=='base_footprint'}.transform.rotation.y`

or 

`/base_footprint.transforms[:].transform.rotation.y`

jhurliman, 2022-03-15:
Changing to a bug since `child_frame_id` can never match against `0`

defunctzombie, 2022-03-15:
Should we remove these suggestions entirely? What determines that the suggestions to filter by a field value like `child_frame_id=="foo"` should be included in the suggestions vs all the other ways we can have message path input values?

jhurliman, 2022-03-15:
> Should we remove these suggestions entirely?

I vote no, I find these suggestions helpful and would not be able to put together the correct syntax by memory without it.

jtbandes, 2022-03-16:
> What determines that the suggestions to filter by a field value like `child_frame_id=="foo"` should be included in the suggestions vs all the other ways we can have message path input values?

Heuristic based on the `_id` suffix

defunctzombie, 2022-03-16:
> I vote no, I find these suggestions helpful and would not be able to put together the correct syntax by memory without it.

My main complaint is they crowd the autocomplete. Maybe if you start typing a `{` then they make sense - but otherwise they feel like lots of noise.

defunctzombie, 2022-04-04:
We need to take the type into account for the autocomplete suggestion.

PR #3190 (2022-04-26): Improve message path autocomplete logic
**User-Facing Changes**
This improves message path autocomplete suggestions.

**Description**
Currently we offer unhelpful message path autocomplete suggestions across arrays of message objects when they contain fields that we mistakenly identify as numeric ids. For example trying to treat the `child_frame_id` field of a transform as a numeric id. The solution in this PR is to check the datatype of that `*_id` field and only offer autocomplete values of `*_id==0` if `*_id` is a primitive integer type.

Fixes #2997
closed 2022-04-26 by foxymiles