foxglove/studio #2999: Set receive time to header stamp time in sample bag

amacneil, 2022-03-09:
**Description**
Our sample bag is currently not particularly user-friendly with some topics such as `/imu` which have infrequent receive times. This leads to unexpected rendering in the Raw Message and Plot panels which default to showing messages by receive time.

We should update the sample bag to pin receive time to header stamp, so that it leads to a more predictable experience for new users.

defunctzombie, 2022-04-07:
@amacneil do you still want this? I see it's gone through some assignment changes but we've never actually thrown it into a project or planning session.

foxymiles, 2022-04-07:
I've made some changes to the https://github.com/foxglove/python-mcap-ros1-support library to make writing a script to do this easier but @jhurliman said there are some other obstacles in the data to making this happen.

jhurliman, 2022-04-08:
Yeah, this is still in progress. The script has to handle MarkerArray and ImageMarkerArray, and I haven't had time this week to touch this.

defunctzombie, 2022-04-08:
Sounds good. I've thrown it on the Q2+ project board so we make some time to push this over the finish line within the quarter.

amacneil, 2022-04-08:
> @ github. com>

I still think this is important, our sample dataset currently gives a bad plotting experience out of the box.

foxymiles, 2022-04-20:
Note from @jhurliman:

```
if you have time today, could you modify it to also check messages 
for a .markers field that is a list, and if the list is non-empty 
use .header.stamp from the first list entry? that will cover 
MarkerArray and ImageMarkerArray which are the last two holdouts
```

foxymiles, 2022-04-20:
Sent convert script to @jhurliman

defunctzombie, 2022-05-14:
What's the status of this? Its been assigned for a while with no activity.

foxymiles, 2022-05-16:
Here's the current script. When I convert nuscenes and play the converted file it looks a bit strange to me like parts of the scene are out of sync with each other. @jhurliman Does this look like it should be doing what we expect?

```
from mcap_ros1.writer import Writer as Ros1Writer
from std_msgs.msg import String  # type: ignore
import rosbag  # type: ignore
import sys

bag = sys.argv[1]
mcap = open("converted.mcap", "w+b")
writer = Ros1Writer(mcap)
for topic, message, msg_time in rosbag.Bag(bag).read_messages():
    if topic == "/tf" and message.transforms:
        message_time = message.transforms[0].header.stamp.to_nsec()
    elif hasattr(message, "markers") and message.markers[0].header.stamp:
        message_time = message.markers[0].header.stamp.to_nsec()
    else:
        message_time = (
            message.header.stamp if message._has_header else msg_time
        ).to_nsec()
    writer.write_message(message=message, topic=topic, log_time=message_time)
```

jhurliman, 2022-05-16:
That script looks right, I wonder what is going wrong?

defunctzombie, 2022-08-02:
@james-rms has taken over nuscenes2mcap and is making some sample data with that. I am renaming this ticket to "update sample data to use mcap" and we should switch over to using the new mcap files from nuscenes (with ros1 messages is fine). Those have the fixed timestamps.

@james-rms once you have a nuscenes.mcap file you are happy with, I can upload it to our assets bucket and update studio to point at that. I am assigning this issue to myself to do that.

defunctzombie, 2022-08-23:
This is blocked on getting a nuscenes file with the duplicate transforms issue fixed.

defunctzombie, 2022-09-26:
Waiting on last few foxglove messages to be protobuf

defunctzombie, 2022-10-13:
@jtbandes Re-assigned to you since you've made the PR to resolve this.

closed 2022-10-13 by jtbandes