foxglove/studio #3004: Figure out how to get Mac app to request permissions to show update notifications

defunctzombie, 2022-03-10:
[image]

jtbandes, 2022-03-10:
I just tested a manual notification with a dev build `new electron.Notification({ title: "Foo bar" }).show();` – this is the same API electron-builder updater uses: https://github.com/electron-userland/electron-builder/blob/17ee450dd747b867c6be5edb153a7fa39f427379/packages/electron-updater/src/AppUpdater.ts#L276

When the `.show()` is called, if the app hasn't been granted permission, this appears immediately:

[image]

When I click Allow, I am able to see the notification that was sent.

[image]

So I think if there is a problem with our auto updater, it's not a permissions problem.

defunctzombie, 2022-03-10:
Probably I accidentally clicked dismiss and didn't realize it ♾️ 

Going to close since it looks like the notifications are working.

defunctzombie, 2022-03-10:
> So I think if there is a problem with our auto updater, it's not a permissions problem.

There's no clear problem other than some folks have said they see notifications and I know I haven't. But its easily because I dismissed the original notice.

closed 2022-03-10 by defunctzombie