foxglove/studio #3008: Table panel should not collapse while playing

amacneil, 2022-03-10:
**Description**
The table panel collapses as soon as you start playback. This makes it very difficult to actually inspect any object.

https://user-images.githubusercontent.com/637671/157745590-f6cdcf94-626a-4ffb-9763-0bce65d527a1.mov

- Version: main
- OS: web

amacneil, 2022-03-12:
@foxymiles is it possible to also fix this for sub-tables? e.g.:

https://user-images.githubusercontent.com/637671/158035478-1047a56f-2b7a-4777-b77f-799776c09b0a.mov

defunctzombie, 2022-03-14:
We do track expanded state in the Raw Message panel and automatically expand any key that we track as expanded. We could adopt a similar pattern here?

What if we drop nested tables as a thing? If you want to view a view that is an array you make that the message path syntax?

foxymiles, 2022-03-14:
Yeah we could probably try to preserve expanded keypaths across messages.

foxymiles, 2022-03-28:
Tracking keypaths actually doesn't seem like a great solution here since it only makes sense if the structure of the message doesn't change. For example in nuscenes the number and order of the markers in the `/markers/annotations` topic changes from message to message so keeping the same keypath expanded across messages isn't really meaningful.

foxymiles, 2022-03-28:
For example if you open nuscenes and a raw messages panel on `/markers/annotations` and expand the 20th row you can see that the item at the 20th row is different from message to message so it doesn't seem like this is really that useful for monitoring data.

defunctzombie, 2022-04-04:
We talked about this and the consensus is that the table panel is targeted at displaying a table of "non-object" values (i.e. an array of "objects" where each field in the object is a primitive value). While the table does support showing a nested table for object values that isn't the target use case so we think that supporting top-level expanded state is sufficient for the purpose of the table panel.

I don't want to remove the nested table feature but if I was to argue for anything that is what I would recommend in the ux here. Rather than a nested table you can open a raw message panel to the value or we can display the object inline.

PR #3014 (2022-03-11): Fix table panel auto-collapse on data change
**User-Facing Changes**
This fixes an issue with the table panel collapsing when it receives new data.

**Description**
This just sets the `autoResetExpanded` flag on the table to false so it doesn't collapse automatically when data changes.

Fixes #3008
closed 2022-04-04 by defunctzombie