foxglove/studio #3041: Deep linking for MCAP files

jameskuszmaul-brt, 2022-03-17:
I would like to be able to generate a URL that I can hand someone that, when they open it, will load in with an MCAP file loaded. This is going to be generated by a script that will also generate the MCAP file, and so in my mind could reasonably take a couple of forms, including:
# Support [deep linking](https://foxglove.dev/docs/studio/deep-linking) for local files. I'm not sure how obnoxious that is to do, but because I know the path on the local machine that the file will be at (and/or can change that path), it is meaningful to be set.
# Add a remote link option for `.mcap` files--I can serve the file on a local web server and let foxglove access it that way.

jtbandes, 2022-03-17:
Thanks for the request!

2. seems like the same request as https://github.com/foxglove/studio/issues/2800. Do you want to keep this ticket open for the deep linking issue, or just close as dupe? I think deep linking to a local file would only work with a desktop app; the browser generally is not able to request access to any file path it dreams up, only access files that the user selects in the open dialog or drags/drops into the page.

jameskuszmaul-brt, 2022-03-17:
I'm happy with closing it as a duplicate. Mostly just wanted to make sure I was asking for the thing I want (making it easy for a user to open a file without extra steps), which might've had some better alternative.

closed 2022-03-17 by jameskuszmaul-brt