foxglove/studio #3053: Move Topic selection out of the panel and into a sidebar

2metres, 2022-03-22:
In order to improve selecting topics and other messages in panels when they are compact move topic selection out of the panel itself and into the panel settings. As per https://github.com/foxglove/ui-playground/pull/4

| 3D Panel | Image Panel |
| --- | --- | 
| [image] |[image]|

amacneil, 2022-03-22:
Did we make any decisions on how we will handle topic settings?

defunctzombie, 2022-03-22:
> Did we make any decisions on how we will handle topic settings?

For the first pass topic settings will continue to open modals.

amacneil, 2022-03-22:
ew, but ok

defunctzombie, 2022-04-15:
> For the first pass topic settings will continue to open modals.

I think this is no longer true with the new 3d panel work.