foxglove/studio #3054: Standardize Panel component/PanelToolbar styles

2metres, 2022-03-22:
## Problem
Currently our panels vary widely in how you interact with them. Visualization panels tend create thier own UI's that overlay the visuals and text based data tend to use the PanelToolbar component. Some toolbars hide and show on hover and create issues for users selecting menus.

## Proposed change
Replace/Refactor the panel to handle the toolbar concept through props.

https://github.com/foxglove/ui-playground/pull/4

[image]

## Challenges
Some panels use the toolbar for various searches. 
Examples: 
- Raw message (message path input), 
- Diagnostics (Panel Filter)
- Log (text search)
- Camera (camera image selection)
- Persistent toolbar takes up more vertical space. 

## Example of current panels
| Panel Name  | |
|---|---|
| 3D Panel | [image] |
| Image Panel |[image] |
| Tab | [image] |
| Log | [image] |
| URDF | [image] |
| Raw message | [image] |
| Diagnostic - Summary | [image] |
| Diagnotics - Detail | [image] |
| State Transition | [image] |
| Plot | [image] |
| Map | [image] |

amacneil, 2022-03-22:
Overall +++ to this effort.

Specifically looking at these icons I have no idea what the first 4 would do. maybe the first one is some sort of menu although its left placement indicates otherwise. No idea on the next 3. Then full screen and settings are obvious (I hope I got them right).

[image]

defunctzombie, 2022-05-11:
https://studio.foxglove.dev/?layoutId=9999af94-710e-4154-bf7b-9af7c7f5aac5

closed 2022-06-06 by defunctzombie