foxglove/studio #3065: Marker delete messages in 3d panel require extraneous frame_id

foxymiles, 2022-03-22:
**Description**
For marker delete messages in the 3d panel to work they seem to require a `frame_id` even though this is not meaningful for simply deleting a marker.

- Version: 1.5.0
- OS: macOS

**Steps To Reproduce**
Send a marker delete message without a `frame_id`.

**Expected Behavior**
It should delete the marker.

**Actual Behavior**
It only works if the message includes a `frame_id`.

defunctzombie, 2022-03-22:
https://github.com/foxglove/studio/blob/main/packages/studio-base/src/panels/ThreeDimensionalViz/SceneBuilder/index.ts#L461-L466

We do not use the frame_id as part of identifying markers for deletion so I think its safe to remove the frame_id requirement.

foxymiles, 2022-03-30:
Looking into this further it looks like the original issue had more to do with message ordering and timestamps and not the frame id. Closing this pending a more specific ticket and example.

closed 2022-03-30 by foxymiles