foxglove/studio #3073: Unreadable mcap file created with python library

defunctzombie, 2022-03-24:
Using the following code snippet and v 0.0.7 of the mcap library I produce example.mcap. When I try to load this mcap file in studio I get the error shown below.

```python
from time import time_ns
import json
from mcap.mcap0.writer import Writer

stream = open("example.mcap", "wb")
writer = Writer(stream)

# The library argument help identify what tool wrote the file.
writer.start(profile="x-custom", library="my-writer-v1")

schema_id = writer.register_schema(
    name="foxglove.FrameTransform",
    encoding="jsonschema",
    data=json.dumps({
        "type": "object",
        "properties": {
            "timestamp": {
                "type": "object",
                "properties": {
                    "sec": {
                        "type": "number"
                    },
                    "nsec": {
                        "type": "number"
                    },
                }
            },
            "parent_frame_id": {
                "type": "string",
            },
            "child_frame_id": {
                "type": "string",
            },
            "transform": {
                "type": "object",
                "properties": {
                    "translation": {
                        "type": "object",
                        "properties": {
                            "x": {
                                "type": "number"
                            },
                            "y": {
                                "type": "number"
                            },
                            "z": {
                                "type": "number"
                            },
                        }
                    },
                    "rotation": {
                        "type": "object",
                        "properties": {
                            "x": {
                                "type": "number"
                            },
                            "y": {
                                "type": "number"
                            },
                            "z": {
                                "type": "number"
                            },
                            "w": {
                                "type": "number"
                            },
                        }
                    },
                }
            },
        }
    }).encode()
)

channel_id = writer.register_channel(
    schema_id=schema_id,
    topic="/tf",
    message_encoding="json",
)

writer.add_message(
    channel_id=channel_id,
    log_time=0,
    data=json.dumps({
        "timestamp": {"sec": 0, "nsec": 0},
        "parent_frame_id": "parent",
        "child_frame_id": "child",
        "transform": {
            "translation": {
                "x": 0,
                "y": 0,
                "z": 0,
            },
            "rotation": {
                "x": 0,
                "y": 0,
                "z": 0,
                "w": 1,
            },
        }
    }).encode('utf-8'),
    publish_time=0,
)

writer.add_message(
    channel_id=channel_id,
    log_time=1000000000,
    data=json.dumps({
        "timestamp": {"sec": 1, "nsec": 0},
        "parent_frame_id": "parent",
        "child_frame_id": "child",
        "transform": {
            "translation": {
                "x": 0,
                "y": 0,
                "z": 0,
            },
            "rotation": {
                "x": 0,
                "y": 0,
                "z": 0,
                "w": 1,
            },
        }
    }).encode('utf-8'),
    publish_time=1000000000,
)

writer.finish()
stream.close()
```

[image]

defunctzombie, 2022-03-24:
@foxymiles @jtbandes I've not investigated whether this is a read-side issue with mcap typescript in studio or a write side issue with the python library. Assigning you both to look into depending on who has time or tell me what I did wrong.

jtbandes, 2022-03-24:
Have you run mcap doctor on the file to see what it says?

I think this issue has occurred in other writers too. Basically some bad logic around updating the messageStartTime when tracking the chunk as it’s written.

jhurliman, 2022-03-24:
Is it possible to convert this into a writer conformance test?

foxymiles, 2022-03-24:
Here's the generated file for reference:

[example.mcap.zip](https://github.com/foxglove/studio/files/8341933/example.mcap.zip)

Mcap doctor flags the error:

```❯ ~/c/mcap/go/cli/mcap/bin/mcap doctor example.mcap 
Examining example.mcap
Chunk.start_time 1000000000 does not match the latest message record time 0
```

defunctzombie, 2022-03-24:
That looks like a bug with the python writer? Or did I add messages incorrectly?

closed 2022-03-24 by foxymiles