foxglove/studio #3074: "Show on startup" checkbox is cut off in open dialog

jtbandes, 2022-03-24:
The checkbox appears cut off, as does the bottom of the "Report a bug" text in the right-hand column:

[image]

amacneil, 2022-04-08:
Upgrading to bug, this is getting worse and is a bad first run experience:

[image]

defunctzombie, 2022-04-08:
@2metres fyi - sending your way

foxymiles, 2022-04-19:
I think this was caused by: https://github.com/foxglove/studio/pull/2563

Maybe setting a min-height here instead of a fixed height is the way to go?

PR #3209 (2022-04-19): Fix open dialog checkbox cutoff
**User-Facing Changes**
This fixes an overflow issue in the open data source dialog.

**Description**
Currently we have a hardcoded height of 520 pixels for this dialog which is not enough to accommodate the content. Rather than set a fixed height, which is fragile. I think we can accomplish the same goal of having a visually stable layout through the steps of the dialog with a min-height.

[image]

Fixes #3074
closed 2022-04-19 by foxymiles