foxglove/studio #3076: Support Robosense lidar

lucabonamini, 2022-03-24:
Any change to get a functionality like the one develop for Velodyne lidar also for Robosense lidar?

jtbandes, 2022-03-25:
What software are you using to interface with the lidar today? For now the easiest way to visualize the scans in Foxglove Studio would be to manually convert them to the `sensor_msgs/PointCloud2` (https://github.com/foxglove/message-schemas/blob/main/proto/ros/sensor_msgs/PointCloud2.proto) and send them over a live connection (e.g. [rosbridge](https://foxglove.dev/docs/studio/connection/rosbridge) or [custom websocket](https://foxglove.dev/docs/studio/connection/foxglove-websocket)).

For scoping: supporting direct connections to Robosense lidars would require custom packet parsing code, akin to https://github.com/foxglove/velodyne-cloud. The RS lidars send data over UDP in a similar way to the Velodyne lidars. They use multiple ports for different purposes (scan data, device info, and configuration). The protocol is described in detail in the user manual. For instance, section 5 in https://lidence.com/wp-content/uploads/2019/07/RS-Lidar-16Users-Guide_v4.0.pdf

lucabonamini, 2022-03-25:
I apologize, my question was not clear. I wanted to know if you are planning to develop an interface for Robosense lidar like [velodyne-cloud](https://github.com/foxglove/velodyne-cloud).

Actually, I use [rslidar_sdk](https://github.com/RoboSense-LiDAR/rslidar_sdk) to interface with the lidar. I think it would be very useful to get visual feedback on sensor data without having to be in ROS ecosystem.

closed 2022-07-08 by lucabonamini