foxglove/studio #3079: Unhelpful data platform error message when user is not signed in

amacneil, 2022-03-25:
**Description**
When user is signed out of the desktop app, and they try to open some data platform data in the desktop app, we currently give them an unhelpful "Status 401" error with a javascript stack trace.

- Version: 1.4.0
- OS: mac

**Current Experience**
[image]

**Desired Experience**
We should tell the user they are not signed in. We should not show a stack trace under any circumstances for network errors like this, they add no additional context and only serve to confuse the user and make our app look buggy.

jtbandes, 2022-03-25:
Maybe related: https://github.com/foxglove/studio/issues/1917

closed 2022-03-30 by jtbandes

jtbandes, 2022-03-30:
@amacneil please try again now that #3093 is merged and see what you think.

The stack trace is still visible. I don't feel great about doing work to intentionally hide stack traces; it feels like it will just make debugging harder in the future.

amacneil, 2022-04-04:
It's definitely an improvement (although the error text is missing some top padding, cc @2metres)

[image]

I think we could still do a lot better though - e.g. there is no link/button to help me with signing in. In this particular error it's probably sensible to either redirect people straight to a login page, or show a modal with signin button. If that's hard to wire up, at least putting a signin button in the data source sidebar would help people move to the next step. 

Hard disagree on showing the stack trace. Stack traces are for unexpected exceptions, and even then they should probably only show in the developer console not our UI. Expected error cases such as "user not logged in" do not deserve a stack trace.

PR #3093 (2022-03-30): Improve error messages for console/DP API auth errors
**User-Facing Changes**
Improved error messages when attempting to access data from Data Platform when not logged in or logged in to the wrong organization.

**Description**
Fixes #3079

Added a custom human-readable message to 401/403 errors.

[image]

Pipe through the `{"error":"..."}` message from the error response in other cases (we were already passing through a `message` field, but not an `error` field).

[image]
commit a9dd0baa (2022-03-30): Improve error messages for console/DP API auth errors (#3093)

**User-Facing Changes**
Improved error messages when attempting to access data from Data Platform when not logged in or logged in to the wrong organization.

**Description**
Fixes #3079

Added a custom human-readable message to 401/403 errors.

Pipe through the `{"error":"..."}` message from the error response in other cases (we were already passing through a `message` field, but not an `error` field).