foxglove/studio #3086: Stamps nsec field missing leading 0

clementchef, 2022-03-29:
**Description**
Small visualization issue. One frequent source of error when looking at timestamps is the lack of leading 0 in the nanoseconds. when you see a number like this (see nsec field in image) you think its toward the end of the second but its .09 not .9.

[image]

- Version: All
- OS: All

**Steps To Reproduce**

**Current Experience**

**Desired Experience**

defunctzombie, 2022-03-30:
I looked into this and have the following thoughts:

One job of the raw message panel is to display raw values. `sec` and `nsec` are the raw "subfields" of a stamp so displaying them in their number form is somewhat "correct" for purposes of debugging the raw message values.

For ROS messages tho the `stamp` field has a `time` type which is mostly treated as a builtin type and the "subfield" serialization is an implementation detail so there's also a notion that the "correct" display should be to just show one value of `seconds.nanoseconds`. 

It's also certainly useful to be able to look at the seconds and nanoseconds values as one numeric unit with nanoseconds being the decimal component of seconds. The "pretty" display next to the _stamp_ field is suppose to provide that functionality but I think its not so obvious how to copy that field data (I could't figure it out when messing with the raw message panel), nor how I might see a decimal seconds values like `1646343346.099993467` and be able to copy that.

One thing we do in some other panels is to follow the "timestamp display" app setting which you can toggle between human time or unix epoch time (seconds.nanoseconds). Would it be useful if we made the raw message panel pretty display next to the stamp field follow the app "timestamp display" setting and be easier to copy? You would see `1646343346.099993467` rather than the `2022-03-3...etc` label and be able to copy the value from that pretty display?

jtbandes, 2022-03-30:
> displaying them in their number form is somewhat "correct" for purposes of debugging the raw message values

Just want to add that IMO a number with leading zeros is still in number form.

PR #3633 (2022-07-02): Pad nsec field with 0's in RawMessage panel
**User-Facing Changes**
Pad `nsec` fields with leading 0's in RawMessage panel.

[image]

**Description**
This helps when visually scanning time values from `sec` and `nsec` fields. A nanosecond label of 099999999 makes it easier to realize this is 0.09 seconds compared to 99999999 which requires some counting to realize this is also 0.09.

Fixes: #3086

Note: This PR does not change diff mode rendering or behavior. No leading 0's are present in diff mode.
[image]
commit 4ce9824d (2022-07-02): Pad nsec field with 0's in RawMessage panel (#3633)

**User-Facing Changes**
Pad `nsec` fields with leading 0's in RawMessage panel.

**Description**
This helps when visually scanning time values from `sec` and `nsec` fields. A nanosecond label of 099999999 makes it easier to realize this is 0.09 seconds compared to 99999999 which requires some counting to realize this is also 0.09.

Fixes: #3086
closed 2022-07-02 by defunctzombie