foxglove/studio #3088: +d

horverno, 2022-03-29:
**Description**

- Version:
- OS:

**Steps To Reproduce**

**Expected Behavior**

**Actual Behavior**

jtbandes, 2022-03-29:
Hi, can you please share more details on the error messages from the developer console? (Press Alt key > View > Advanced)

It seems like the problem might be that the `nissan_bringup` package is not in your `ROS_PACKAGE_PATH` (that's why you get the "404" errors when trying to load the meshes). You can edit the `ROS_PACKAGE_PATH` by updating the environment variable before running Studio (e.g. launching Studio after `source ~/catkin_ws/devel/setup.bash`), or directly in the app settings (bottom button in the sidebar).

horverno, 2022-03-29:
Hi @jtbandes 

Thanks for the quick reply, i tried your suggestion sadly it did not worked.

The errors are the following:

```
x-foxglove-ros-package:?targetPkg=nissan_bringup&relPath=param%2Fnissan_leaf_robotdescription.ref.xml:1          

Failed to load resource: net::ERR_<unknown>
```

```
TypeError: Failed to fetch: x-foxglove-ros-package:?targetPkg=rviz_markers&relPath=stl%2Flexus.stl
```

Can I use an absolute path instead of the relative one somehow? Maybe this could solve the issue.

jtbandes, 2022-03-29:
Can you also try with Studio v1.5? Some changes to package loading were made which might affect this: https://github.com/foxglove/studio/pull/2970

horverno, 2022-03-30:
I learned that the newest build available is 1.4.0 at the moment.

``` r
$ sudo apt install foxglove-studio
Reading package lists... Done
Building dependency tree       
Reading state information... Done
foxglove-studio is already the newest version (1.4.0).
```

In the meantime I also tested it on `Ubuntu 20.04` and `ROS noetic` and `Ubuntu 18.04` and `ROS melodic`, the same issue is present.

I have two options now: wait for `v1.5` or build from source which I did not found any instructions. Considering the speed of the releases it only will take a couple of days, so I decided to wait.

horverno, 2022-03-31:
hi @jtbandes  I tried `v1.5.1` different error message, but same output

[image]

defunctzombie, 2022-04-01:
@horverno Thanks for following up with this. To help us reproduce the issue - could you confirm what your ROS_PACKAGE_PATH setting is within the Studio desktop app? I see above that you clone the repos to `~/catkin_ws/src` and want to make sure I've understood correctly how that relates to the package path.

horverno, 2022-04-01:
@defunctzombie thanks, i can confirm, that the ROS_PACKAGE_PATH in the Foxglove Studio desktop app contains among others the following: 
``` r
/home/he/catkin_ws/src/nissan_leaf_ros/nissan_bringup:/home/he/catkin_ws/src/rviz_markers:/opt/ros/melodic/share
```

[image]

horverno, 2022-04-11:
@defunctzombie @jtbandes i tried `v1.6.0` but the issue is still there

I also ask the people who i know have experience with this: @rudolfkrecht @norbertmarko @umiklos @csaplaci where somebody able to load 3d models? If yes, how?

defunctzombie, 2022-04-11:
@horverno the referenced fix is not in a release yet.

Also worth noting that your ROS_PACKAGE_PATH has `/home/he/catkin_ws/src/nissan_leaf_ros/nissan_bringup` which isn't correct. It should be `/home/he/catkin_ws/src` to point to the root where to find packages.

Until the fix is part of a release I think you could try `/home/he/catkin_ws/src/nissan_leaf_ros` as a path. With the fix, you will be able to specify the workspace root src and Studio will find packages recursively from there.

horverno, 2022-04-11:
thanks @defunctzombie 

did I understand correctly that your suggestion was to directly give `/robot_description` the path like this: 

```
/home/he/catkin_ws/src/nissan_leaf_ros/nissan_bringup/param/nissan_leaf_robotdescription.ref.xml
```
instead of `package://` like this
```
package://nissan_bringup/param/nissan_leaf_robotdescription.ref.xml
```

[image]

I did this but did not solve my issues. Also I tried other URDF's such as [ANYbotics/anymal_c_simple_description](https://github.com/ANYbotics/anymal_c_simple_description) these won't work either.

horverno, 2022-04-12:
@defunctzombie @jtbandes i tried `v1.7.0` the error message disappeared, but still no 3D model

``` csharp
[main/main.js] Foxglove Studio 1.7.0
[main/main.js] initializing Sentry in main
[main/main.js] Could not set app as handler for foxglove://
[main/main.js] Elapsed (ms) until new StudioWindow: 123
[main/main.js] New Foxglove Studio window 1
[main/main.js] Starting update loop
[main/main.js] load-pending-files
[main/main.js] injectFilesToOpen: no files - skipping
[main/main.js] window URL loaded
[main/main.js] Load: package://nissan_bringup/param/nissan_leaf_robotdescription.ref.xml
[main/main.js] ROS_PACKAGE_PATH /home/he/catkin_ws/src/nissan_leaf_ros
[main/main.js] searching /home/he/catkin_ws/src/nissan_leaf_ros for packages
[main/main.js] Resolved: /home/he/catkin_ws/src/nissan_leaf_ros/nissan_bringup/param/nissan_leaf_robotdescription.ref.xml
[main/main.js] Checking for updates
```

defunctzombie, 2022-04-16:
@horverno Thanks for following up. I've looked into this and determined that the root cause is that we don't _yet_ support loading `.obj` meshes. We are actively working on many 3d panel enhancements that will allow us to better support meshes like `.obj` files.

I'm going to change this ticket from a bug to a feature and update the title to indicate that the 3d panel should support .obj meshes for URDF (robot descriptions) and mesh markers.

In the meantime, I've also made a ticket to improve the error feedback when we do encounter a mesh we don't yet support. This should not fail silently but should let you know what happened.

https://github.com/foxglove/studio/issues/3198

horverno, 2022-04-20:
I can confirm that with `v1.8.0` the situation is better. Our URDF still  have some issues, but https://github.com/ANYbotics/anymal_b_simple_description seem to work. We will compare what's the difference and will improve accordance to that. Thanks
[image]
[image]
.

jhurliman, 2022-07-14:
The title of this ticket, ".obj mesh support", is completed in the new `3D (Beta)` panel. If there are additional rendering issues with the URDF support in the new 3d panel in 1.18 (due to release early next week) let's open a new ticket to scope down the work and classify it as a bug report or feature request.

PR #3157 (2022-04-08): Recursively search ROS_PACKAGE_PATH in findRosPackage

**User-Facing Changes**
Users can follow ROS conventions and set the ROS_PACKAGE_PATH to the root of their workspace. Studio (desktop only) will traverse this path to find all packages within it when resolving `package://` urls.

**Description**
ROS package:// urls follow the convention of `package://<package name>/<path to file>`. In ROS projects, it is common to have a monorepo with several packages folders. This folder would be in the ROS workspace. When setting the ROS_PACKAGE_PATH environment variable or app setting to point to the wo [... cut]
commit b2f2597d (2022-04-08): Recursively search ROS_PACKAGE_PATH in findRosPackage (#3157)

ROS package:// urls follow the convention of `package://<package name>/<path to file>`. In ROS projects, it is common to have a monorepo with several packages folders. This folder would be in the ROS workspace. When setting the ROS_PACKAGE_PATH environment variable or app setting to point to the workspace, the correct behavior for finding packages is to look recursively at the ROS_PACKAGE_PATH location and any locations below it to find the package.

This change updates our package lookup logic to use a recursive search on the ROS_ [... cut]
closed 2022-07-14 by jhurliman