foxglove/studio #3090: Plotting 2-D time-series data from chunked MCAP file has extra lines

jameskuszmaul-brt, 2022-03-29:
**Description**

If I create an indexed MCAP file with multiple chunks and then attempt to load it in Studio, either via URL or local file, I end up with a bunch of extraneous lines polluting the time-series plot. E.g., while still loading some chunked data I had:

[image]

It looks like points at the chunk boundary are getting tied to non-adjacent points for some reason. If I make it so that the MCAP file is just one big chunk, then it is fine, but that feels a bit silly.

Additionally, and this may be unrelated, it generally looks like the sub-sampled lines still show their connections even once the detailed points are filled in.

**Steps To Reproduce**

Attempt to create a 2-D plot of a signal in a MCAP file that has multiple chunks.

**Expected Behavior**

Lines are only drawn between adjacent points. Will add a MCAP file + relevant signal to this ticket for reproduction purposes.

**Actual Behavior**

Too many lines.

jameskuszmaul-brt, 2022-03-29:
Reproduction MCAP file with many chunks:
[many_chunks.mcap.gz](https://github.com/foxglove/studio/files/8373403/many_chunks.mcap.gz)

Exactly the same data, but just one chunk:
[one_chunk.mcap.gz](https://github.com/foxglove/studio/files/8373408/one_chunk.mcap.gz)

Look at the `"/imu/aos aos.timing.Report".pid` signal, for instance (see, e.g., this layout: 
```json
{
  "configById": {
    "Plot!4ij8lha": {
      "paths": [
        {
          "value": "\"/imu/aos aos.timing.Report\".pid",
          "enabled": true,
          "timestampMethod": "receiveTime"
        }
      ],
      "minYValue": "",
      "maxYValue": "",
      "showXAxisLabels": true,
      "showYAxisLabels": true,
      "showLegend": true,
      "legendDisplay": "floating",
      "showPlotValuesInLegend": false,
      "isSynced": true,
      "xAxisVal": "timestamp",
      "sidebarDimension": 240
    },
    "SourceInfo!3mzaul8": {}
  },
  "globalVariables": {},
  "userNodes": {},
  "linkedGlobalVariables": [],
  "playbackConfig": {
    "speed": 1,
    "messageOrder": "receiveTime"
  },
  "layout": {
    "first": "Plot!4ij8lha",
    "second": "SourceInfo!3mzaul8",
    "direction": "row",
    "splitPercentage": 70.79067472554583
  }
}
```

jtbandes, 2022-03-29:
Running `mcap doctor` on the file, I get this:
```
Examining /Users/jacob/Downloads/foo-3.mcap
Chunk.start_time 137401467398 does not match the latest message record time 140401410174
Chunk.start_time 137401467398 does not match the latest message record time 143401407969
Chunk.start_time 137401467398 does not match the latest message record time 146401406300
Chunk.start_time 137401467398 does not match the latest message record time 149401408095
Chunk.start_time 137401467398 does not match the latest message record time 152401420852
Chunk.start_time 137401467398 does not match the latest message record time 155401405239
Chunk.start_time 137401467398 does not match the latest message record time 158401373330
Chunk.start_time 137401467398 does not match the latest message record time 161401407569
Chunk.start_time 137401467398 does not match the latest message record time 164401482697
Chunk.start_time 137401467398 does not match the latest message record time 167401407362
Chunk.start_time 137401467398 does not match the latest message record time 170401426582
Chunk.start_time 137401467398 does not match the latest message record time 173401408692
```
(Note the message is incorrect — it should say "does not match the earliest message log time")

The first thing I would try is addressing that issue in your conversion tool and see if that helps.

jtbandes, 2022-03-29:
The ChunkIndex records at the end of the file don't match the messageStart/EndTime from the chunks themselves, so these also need to be updated. I will work on enhancing the doctor tool to catch this error too.

jameskuszmaul-brt, 2022-03-29:
That seems to have done it. Thanks!

closed 2022-03-29 by jameskuszmaul-brt

jtbandes, 2022-03-29:
https://github.com/foxglove/mcap/pull/350

amacneil, 2022-03-29:
Given that mcap is a pretty new technology, should we add some of these "mcap doctor" checks into Studio so that we can warn people if they are opening potentially incorrect files?

jameskuszmaul-brt, 2022-03-29:
I suspect just having doctor advertised a bit better might help. When doing this, I had already run into a couple of error messages in studio due to having inconsistent logs, and for whatever reason I just hadn't fully realized that doctor was available--had there been any suggestion to run `mcap logger` at any point, I might've done it. `doctor` also doesn't appear to be mentioned either in the top-level mcap README nor in the [file format specification](https://github.com/foxglove/mcap/tree/main/docs/specification)

jtbandes, 2022-03-30:
https://github.com/foxglove/mcap/pull/353