foxglove/studio #3092: Failed to load DAE files using data uri

gonzalocasas, 2022-03-29:
**Description**

I tried publishing the meshes of a robot as visualization markers (`visualization_msgs/Marker`) to a topic, which works for STL files but not DAE files. My robot model is the UR3e, directly off of the official ROS repos. The STL files are collision geometry, and the DAE files are visual geometry.

- Version: https://studio.foxglove.dev/
- OS: Windows 10, Chrome latest

**Steps To Reproduce**

The visual meshes are published to a topic using data URIs of the following format: `data:model/vnd.collada+xml;base64,...`. For STL meshes (which work), the data URI is `data:application/sla;base64,...`, so it seems there's a problem with data type identification.

To reproduce:
 - Launch ROS+MoveIt+UR3e using [this docker compose file](https://github.com/compas-teaching/COMPAS-II-FS2022/blob/main/docker/moveit-noetic/docker-compose.yml)
 - Using [`compas_fab`](https://pypi.org/project/compas-fab/) in a Python environment, run [this script](https://gist.github.com/gonzalocasas/a9c49a0f5359873976eda4c9ba1e95c5)

**Expected Behavior**

Loaded meshes.

**Actual Behavior**

[image]

gonzalocasas, 2022-03-29:
/cc @mattiskoh

jtbandes, 2022-03-29:
Thank you for the report! I'm going to convert this to a feature request since it's more of a missing feature within the app. We currently have a MIME type check for STL files but not DAE:

https://github.com/foxglove/studio/blob/34f8298432af3268eb671f96768b9914b41686b4/packages/studio-base/src/panels/ThreeDimensionalViz/commands/MeshMarkers.tsx#L65-L73

jhurliman, 2022-03-30:
@gonzalocasas this should be working now, you can try it out on <https://studio.foxglove.dev/>. Please reopen or create a new ticket if you are running into any issues.

closed 2022-03-30 by jhurliman

gonzalocasas, 2022-04-01:
Yay! That works! Thanks a lot @jhurliman !

The meshes load correctly now. I am not yet sure how to set them to their correct frame/transformation thou (maybe not a question for this issue), if I set `frame_id` of the header to be the link name of each mesh, they appear somewhat in position (but some weird orientation I think), but then disappear a few seconds later because there's a missing transform between each link's frame and `base_link`, however, the frame tree seems to be correct (it also works correctly on moveit), so, I'm not sure what I am setting incorrectly:
[image]

jhurliman, 2022-04-01:
@gonzalocasas I converted your question to a discussion at https://github.com/foxglove/studio/discussions/3139