foxglove/studio #3094: Studio publisher doesn't re-connect after roscore restarts (ros1 live connection)

jessicaaustin, 2022-03-29:
When using a panel that publishes messages, if the underlying ROS connection is lost and regained, then the panel does not automatically re-connect to subscribers. Reloading the layout or restarting studio will re-connect the studio publisher to its subscriber(s). 

This happens both via the Extensions API and the Publish panel.

- Version: Studio 1.4.0
- OS: Ubuntu (but i don't think this matters)
- ROS 1, live connection, all on one host

**Steps To Reproduce**

See also https://github.com/foxglove/studio/issues/2950 -- it's the same setup, just a different scenario.

1. In one terminal, start `roscore`
1. In another terminal, run `rostopic echo /chatter`
2. Open Studio and add a Publish panel, and configure it to publish to `/chatter` with message type `std_msgs/String`
3. Send a message via Studio -- you will see it in the `rostopic echo` terminal
4. Restart `roscore`, and the `rostopic echo /chatter` processes, but _do not restart Studio or reload the layout_
5. Attempt to send a message via Studio

**Expected Behavior**

* The message is received by the  `rostopic echo /chatter` subscriber

**Actual Behavior**

* Nothing shows up in the  `rostopic echo /chatter` terminal
* There are no errors/warnings in Studio
* If you restart studio or reload the layout, or open a new Publish panel, then it connects to the  `rostopic echo /chatter` subscriber and publishes messages

**Screenshots**

Successfully sending messages:

[image]

Restart roscore, and try again. Note how the message is not sent to any subscribers, either the `rostopic echo` or the `Raw Messages` panel. The error shows up when you restart roscore, but nothing shows up when you hit the Publish button.

[image]

jessicaaustin, 2022-03-29:
Obviously this issue has a clear workaround, but it would be nice to see it fixed -- it's especially confusing because the Studio _subscribers_ manage to re-connect after roscore restarts, so you see data flowing through again for most panels. But then trying to publish fails with no errors.

jhurliman, 2022-04-08:
I believe this is standard ROS behavior. I tried the following repro steps:

1. In one terminal, start `roscore`
2. In another terminal, run `rostopic echo /chatter`
3. In another terminal, run `rostopic pub -r 1 /chatter std_msgs/String '{data: "test"}'` and confirm the echo terminal sees one message per second
4. Restart `roscore` and the `rostopic echo /chatter` processes, but do not restart the publisher
5. Observe `WARNING: topic [/chatter] does not appear to be published yet` and no messages arrive in the new echo process

When the roscore process stops and a new one starts, there is no communication between the publisher and the roscore server (in either direction) that would tell the publisher it needs to re-advertise its published topics. So any unbroken TCP connections between nodes will continue to operate, but if a new node starts up and asks the new roscore process for a list of published topics, it will get an empty list back. This fragility in roscore was fixed in ROS2 by moving to a peer-to-peer discovery process.

jessicaaustin, 2022-04-12:
@jhurliman Thanks, that makes sense. And I agree there are ROS1 limitations here that are outside Studio's control.

The main issue here is notifying the user that the panel they're using to publish messages is no longer working. Ideally, the panel could also recover from this state, but at the very least if the user is notified then they can reload the panel themselves. I'm trying to figure out a mechanism within my custom Extension where I can do this.

For example, say I decide to listen to a `/heartbeat` topic within my extension, and use that to detect if the ROS connection was lost or unstable. That doesn't quite work, because depending on how things were restarted, maybe the studio subscribers would reconnect, but the _publisher_ would not.

I'm not aware of any mechanism [via the Panel API](https://foxglove.dev/docs/studio/extensions/panel-api-methods) where I could detect that my publisher is no longer connected to any subscribers. It seems like that would be the ideal information to use to notify the user that the message they're publishing isn't actually going anywhere.

So I'm going to close this ticket and create a new one, to see if there is a way to expose more information about the ROS connection to the Extension.

closed 2022-04-12 by jessicaaustin