foxglove/studio #3096: Studio publisher does not connect to C++ subscriber (ROS1 live connection)

jessicaaustin, 2022-03-29:
When publishing messages via Studio, python subscribers can receive the messages no problem. But a subscriber written using C++ does not receive messages.

- Version: Studio 1.4.0
- OS: Ubuntu (but i don't think this matters)
- ROS 1, live connection, all on one host

**Steps To Reproduce**

1. Create a simple ROS subscriber in C++ 
   * I just used the one from the ROS tutorials: [Writing the Subscriber Node](http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28c%2B%2B%29)
   * Any time it receives a message on `/chatter`, it writes it as a WARN message to the log
2. Launch the subscriber node
3. Verify the subscriber works -- run `rostopic pub  /chatter std_msgs/String "data: 'hi from rostopic pub'"` and check that the message shows up in the log
4. Launch Studio
5. Attempt to send a message via Studio

**Expected Behavior**

* The message is received by the C++ subscriber, and shows up in the log.

**Actual Behavior**

* No message is received
* There is an error in the node log: `Failed to contact publisher [hostname:43047] for topic [/chatter]` -- this is the foxglove studio publisher 

**Screenshots and logs**

[image]

Full node log:

```
1648577826.702250150  Node Startup
1648577826.710711478 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/subscription.cpp:244(Subscription::pubUpdate)] [topics: /rosout] Publisher update for [/chatter]:  already have these connections: 
1648577826.930959538 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/transport/transport_tcp.cpp:645(TransportTCPPtr ros::TransportTCP::accept)] [topics: /rosout] Accepted connection on socket [7], new socket [11]
1648577826.931075062 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/transport/transport_tcp.cpp:142(TransportTCP::initializeSocket)] [topics: /rosout] Adding tcp socket [11] to pollset
1648577826.931091639 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/connection_manager.cpp:188(ConnectionManager::tcprosAcceptConnection)] [topics: /rosout] TCPROS received a connection from [127.0.0.1:57050]
1648577826.931134390 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/connection_manager.cpp:203(ConnectionManager::onConnectionHeaderReceived)] [topics: /rosout] Connection: Creating TransportSubscriberLink for topic [/rosout] connected to [callerid=[/rosout] address=[TCPROS connection on port 45159 to [127.0.0.1:57050 on socket 11]]]
1648577829.534235544 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/topic_manager.cpp:561(TopicManager::pubUpdate)] [topics: /rosout] Received update for topic [/chatter] (1 publishers)
1648577829.534251940 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/subscription.cpp:244(Subscription::pubUpdate)] [topics: /rosout] Publisher update for [/chatter]: http://mimir:44265/,  already have these connections: 
1648577829.534336646 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/subscription.cpp:421(Subscription::negotiateConnection)] [topics: /rosout] Began asynchronous xmlrpc connection to [mimir:44265]
1648577829.734995483 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/subscription.cpp:509(Subscription::pendingConnectionDone)] [topics: /rosout] Connecting via tcpros to topic [/chatter] at host [mimir:35493]
1648577829.735163047 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/transport/transport_tcp.cpp:326(TransportTCP::connect)] [topics: /rosout] Resolved publisher host [mimir] to [127.0.1.1] for socket [13]
1648577829.735217721 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/transport/transport_tcp.cpp:142(TransportTCP::initializeSocket)] [topics: /rosout] Adding tcp socket [13] to pollset
1648577829.735234714 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/transport/transport_tcp.cpp:365(TransportTCP::connect)] [topics: /rosout] Async connect() in progress to [mimir:35493] on socket [13]
1648577829.735284236 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/subscription.cpp:525(Subscription::pendingConnectionDone)] [topics: /rosout] Connected to publisher of topic [/chatter] at [mimir:35493]
1648577829.736062366 WARN /chatter_listener [/home/jessica/dev/personal/ros_tutorials/catkin_ws/src/beginner_tutorials/src/listener.cpp:6(chatterCallback)] [topics: /rosout] I heard: [hi from rostopic pub]
1648577837.847345296 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/topic_manager.cpp:561(TopicManager::pubUpdate)] [topics: /rosout] Received update for topic [/chatter] (2 publishers)
1648577837.847374225 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/subscription.cpp:244(Subscription::pubUpdate)] [topics: /rosout] Publisher update for [/chatter]: http://mimir:44265/, http://mimir:43047/,  already have these connections: http://mimir:44265/, 
1648577837.847490289 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/subscription.cpp:421(Subscription::negotiateConnection)] [topics: /rosout] Began asynchronous xmlrpc connection to [mimir:43047]
1648577838.049727037 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/xmlrpc_manager.cpp:208(XMLRPCManager::validateXmlrpcResponse)] [topics: /rosout] XML-RPC call [requestTopic] didn't return an array
1648577838.049748993 DEBUG /chatter_listener [/tmp/binarydeb/ros-melodic-roscpp-1.14.10/src/libros/subscription.cpp:471(Subscription::pendingConnectionDone)] [topics: /rosout] Failed to contact publisher [mimir:43047] for topic [/chatter]
```

Topic info:

```
mimir:~$ rostopic info /chatter
Type: std_msgs/String

Publishers: 
 * /rostopic_31441_1648577829324 (http://mimir:44265/)
 * /foxglovestudio_31584 (http://mimir:43047/)

Subscribers: 
 * /chatter_listener (http://mimir:40991/)
 * /foxglovestudio_31584 (http://mimir:43047/)
```

jtbandes, 2022-03-29:
If you are using two nodes at once for testing, could this be the same issue as https://github.com/foxglove/ros1/pull/19 ?

jessicaaustin, 2022-03-29:
Good question. Tried again with single publisher and subscriber:

```
mimir:~$ rostopic info /chatter
Type: std_msgs/String

Publishers: 
 * /foxglovestudio_20554 (http://mimir:39097/)

Subscribers: 
 * /chatter_listener (http://mimir:39663/)

```

Same result -- no messages received by the cpp node, and the same error message in the log.

jhurliman, 2022-04-08:
I can reproduce this. Investigating now.

jhurliman, 2022-04-08:
It looks like roscpp does not support `Transfer-Encoding: chunked`. Fixing this in [@foxglove/xmlrpc](https://github.com/foxglove/xmlrpc/pull/12)

jessicaaustin, 2022-04-12:
@jhurliman Thanks for looking into this. 

I just downloaded Studio 1.7.0 to test this out. But I am still seeing the same issue. What am I missing? Was there something different about your test setup maybe?

[image]

working fine for a python node:

[image]

jhurliman, 2022-04-13:
@jessicaaustin uh oh! Let me try to reproduce your environment as closely as possible. Can you let me know what version of ROS1 (Noetic?) you are using, and which version of Ubuntu?

jessicaaustin, 2022-04-13:
No, it's melodic, on ubuntu 18.04 :sweat:

jhurliman, 2022-04-19:
@jessicaaustin can you try the foxglove/ros1 repo to see if this issue reproduces there with a more minimal example?

Instructions:
1. git clone the https://github.com/foxglove/ros1 repo
2. `yarn && yarn build`
3. `cd examples/ros1-chatter`
4. `yarn`
5. (start roscore in another tab if it's not already running)
6. `yarn start`

This will start a simple node that publisher to the `/chatter` topic. Next, start your own C++ node that subscribes to `/chatter` and see if it works or if you encounter the same issue.

jessicaaustin, 2022-04-22:
@jhurliman Yes, this does cause an error:

[image]

I'm going to see if I can reproduce this running inside of a docker container instead of just running on my machine directly. Then we can compare apples to apples. I assume you're doing a similar thing (VM or docker) for your own testing?

jessicaaustin, 2022-04-22:
@jhurliman ok check this out:

https://github.com/jessicaaustin/ros_demo

Running using those instructions, I can replicate using either melodic or noetic:

[image]

Let me know if this helps!

jhurliman, 2022-04-22:
@jessicaaustin interesting, I can't reproduce this using your demo repo and the latest @foxglove/ros1. This is what I get (host is a Linux 20.04 VM, using noetic+focal settings for your container):

```
$ docker run --rm -it --net host ros_demo:${ROS_RELEASE} rosrun beginner_tutorials chatter_listener
[DEBUG] [1650660081.065709063] [/listener] [TransportTCP::initializeSocket:143]: Adding tcp socket [7] to pollset
[DEBUG] [1650660081.066282266] [/listener] [TransportUDP::createIncoming:268]: UDPROS server listening on port [45339]
[DEBUG] [1650660081.067474338] [/listener] [start:413]: Started node [/listener], pid [1], bound on [ubuntu-linux-20-04-desktop], xmlrpc port [35545], tcpros port [42033], using [real] time
Warning: recursive print statement has occurred.  Throwing out recursive print.
[DEBUG] [1650660081.068883950] [/listener] [Subscription::pubUpdate:255]: Publisher update for [/chatter]:  already have these connections: 
[DEBUG] [1650660081.361607630] [/listener] [TransportTCPPtr ros::TransportTCP::accept:678]: Accepted connection on socket [7], new socket [11]
[DEBUG] [1650660081.361735587] [/listener] [TransportTCP::initializeSocket:143]: Adding tcp socket [11] to pollset
[DEBUG] [1650660081.361758670] [/listener] [ConnectionManager::tcprosAcceptConnection:188]: TCPROS received a connection from [127.0.0.1:53338]
[DEBUG] [1650660081.361877794] [/listener] [TransportTCP::read:495]: Async socket[11] is connected
[DEBUG] [1650660081.361909919] [/listener] [ConnectionManager::onConnectionHeaderReceived:202]: Connection: Creating TransportSubscriberLink for topic [/rosout] connected to [callerid=[/rosout] address=[TCPROS connection on port 42033 to [127.0.0.1:53338 on socket 11]]]
[DEBUG] [1650660185.282974570] [/listener] [TopicManager::pubUpdate:561]: Received update for topic [/chatter] (1 publishers)
[DEBUG] [1650660185.283096027] [/listener] [Subscription::pubUpdate:255]: Publisher update for [/chatter]: http://ubuntu-linux-20-04-desktop:37565/,  already have these connections: 
[DEBUG] [1650660185.283451524] [/listener] [Subscription::negotiateConnection:432]: Began asynchronous xmlrpc connection to [ubuntu-linux-20-04-desktop:37565]
[DEBUG] [1650660185.494798415] [/listener] [Subscription::pendingConnectionDone:520]: Connecting via tcpros to topic [/chatter] at host [127.0.1.1:43859]
[DEBUG] [1650660185.495348952] [/listener] [TransportTCP::initializeSocket:143]: Adding tcp socket [13] to pollset
[DEBUG] [1650660185.495420534] [/listener] [TransportTCP::connect:366]: Async connect() in progress to [127.0.1.1:43859] on socket [13]
[DEBUG] [1650660185.495528950] [/listener] [TransportTCP::read:495]: Async socket[13] is connected
[DEBUG] [1650660185.495594991] [/listener] [Subscription::pendingConnectionDone:536]: Connected to publisher of topic [/chatter] at [127.0.1.1:43859]
[ WARN] [1650660185.497833471] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660186.244318571] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660187.246963987] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660188.250924058] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660189.253638642] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660190.258307793] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660191.265104468] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660192.270341325] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660193.272065923] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660194.275206844] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660195.277006154] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660196.279218502] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660197.282002472] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660198.287520585] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660199.290809427] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660200.294119022] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[ WARN] [1650660201.298185735] [/listener] [chatterCallback:6]: I heard: [Hello, world!]
[DEBUG] [1650660202.306942282] [/listener] [TransportTCP::read:532]: Socket [13] received 0/4 bytes, closing
[DEBUG] [1650660202.307252279] [/listener] [TransportTCP::close:465]: TCP socket [13] closed
[DEBUG] [1650660202.307310862] [/listener] [Connection::drop:327]: Connection::drop(0)
[DEBUG] [1650660202.307347987] [/listener] [TransportPublisherLink::onConnectionDropped:273]: Connection to publisher [TCPROS connection on port 59936 to [127.0.1.1:43859 on socket 13]] to topic [/chatter] dropped
[DEBUG] [1650660202.307712233] [/listener] [TransportTCP::socketUpdate:748]: getsockopt failed on socket [-1]
[DEBUG] [1650660202.307824066] [/listener] [TransportTCP::socketUpdate:755]: Socket -1 closed with (ERR|HUP|NVAL) events 8193: Unknown error -1
[DEBUG] [1650660202.360268418] [/listener] [TopicManager::pubUpdate:561]: Received update for topic [/chatter] (0 publishers)
[DEBUG] [1650660202.360308585] [/listener] [Subscription::pubUpdate:255]: Publisher update for [/chatter]:  already have these connections: http://ubuntu-linux-20-04-desktop:37565/, 
[... cut ...]
[DEBUG] [1650660202.360363376] [/listener] [Connection::drop:327]: Connection::drop(2)
```

Other terminal:

```
$ yarn start
yarn run v1.22.10
$ ts-node src/index.ts
rosfollower listening at http://ubuntu-linux-20-04-desktop:37565/
publishing /chatter (std_msgs/String)
registered publisher for /chatter (std_msgs/String)
registered as a publisher for /chatter, 1 current subscriber(s)
requestTopic [ '/listener', '/chatter', [ [ 'TCPROS' ] ] ]
[ 1, '', [ 'TCPROS', '127.0.1.1', 43859 ] ]
adding subscriber tcpros://127.0.0.1:59936 (/listener) to topic /chatter, connectionId 0
```

jessicaaustin, 2022-04-22:
@jhurliman Interesting... 

ok to double-check two things:

* is your ros1 repo is at the same commit as me (see below)?
* where is your roscore running in the example above?

my ros1 repo was pulled today and had no changes:

```
mimir:~/dev/foxglove/ros1/examples/ros1-chatter$ git remote -v
origin	https://github.com/foxglove/ros1.git (fetch)
origin	https://github.com/foxglove/ros1.git (push)

mimir:~/dev/foxglove/ros1/examples/ros1-chatter$ git rev-parse HEAD
369f910958a3505112b28a113fa19b268fe3e3bb

mimir:~/dev/foxglove/ros1/examples/ros1-chatter$ git status
On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   yarn.lock

no changes added to commit (use "git add" and/or "git commit -a")
```

jessicaaustin, 2022-04-22:
Yeah, the yarn.lock that changed is the one at https://github.com/foxglove/ros1/blob/main/examples/ros1-chatter/yarn.lock -- in the repo it's at foxglove/xmlrpc@^1.1.9 but it got upgraded to foxglove/xmlrpc@^1.2.0

```
diff --git a/examples/ros1-chatter/yarn.lock b/examples/ros1-chatter/yarn.lock
index fb49957..eaac563 100644
--- a/examples/ros1-chatter/yarn.lock
+++ b/examples/ros1-chatter/yarn.lock
@@ -30,11 +30,11 @@
     fetch-blob "^2.1.1"
 
 "@foxglove/ros1@file:../..":
-  version "1.4.0"
+  version "1.4.2"
   dependencies:
     "@foxglove/rosmsg" "^2.0.0 || ^3.0.0"
     "@foxglove/rosmsg-serialization" "^1.2.3"
-    "@foxglove/xmlrpc" "^1.1.9"
+    "@foxglove/xmlrpc" "^1.2.0"
     eventemitter3 "^4.0.7"
 
 "@foxglove/rosmsg-serialization@^1.2.3":
@@ -51,10 +51,10 @@
   dependencies:
     md5-typescript "^1.0.5"
 
-"@foxglove/xmlrpc@^1.1.9":
-  version "1.1.9"
-  resolved "https://registry.yarnpkg.com/@foxglove/xmlrpc/-/xmlrpc-1.1.9.tgz#15d338f585b538865c0a281d2787a4204721389e"
-  integrity sha512-Lf49vDIh3ewAEtYqI7IeczLfxC4YHRqyavE613KlOoPkBvVrkkjco4lJuFKiSNd0AieJN9D2GV3vZpe5sa2/Cw==
+"@foxglove/xmlrpc@^1.2.0":
+  version "1.2.0"
+  resolved "https://registry.yarnpkg.com/@foxglove/xmlrpc/-/xmlrpc-1.2.0.tgz#90018a21ab2c897cfbc268d74fe9f7401123926a"
+  integrity sha512-zY1uL0sVaT+8Ekn72eN85n0Q5sMbcOYNjyum2B9Qqvh9w1vty5q6HwO0Gg5HJj65lpiGZTjuEzKM4drEMaY2Ng==
   dependencies:
     "@foxglove/just-fetch" "^1.2.4"
     byte-base64 "^1.1.0"
```

jhurliman, 2022-04-22:
@jessicaaustin I just updated my examples/ros1-chatter to use @foxglove/xmlrpc 1.2.0 instead of 1.1.9 and I can reproduce your issue now. Looking into the actual cause now, thank you!

jhurliman, 2022-04-25:
Fix is landing now: https://github.com/foxglove/studio/pull/3259

closed 2022-04-27 by jhurliman

jessicaaustin, 2022-05-13:
Confirmed this does work: both the ros1-chatter example, and publishing via Studio v1.11.0. Thank you! :pray: