foxglove/studio #3099: If log is longer than 1200 sec, can't reliably play/seek while log is loading

jameskuszmaul-brt, 2022-03-29:
**Description**

On a sufficiently long/large logfile, `_readAheadBlocks` will evaluate to 1, since the block size (in time) will be greater than `READ_AHEAD_NS`:
https://github.com/foxglove/studio/blob/0787f9f20d3a01445f92dfc6845a3c898bd25f57/packages/studio-base/src/randomAccessDataProviders/MemoryCacheDataProvider.ts#L295

If you then attempt to play the logfile, we then encounter the below calculation of start/end times, which will attempt to request an end time 1 block into the future:

https://github.com/foxglove/studio/blob/0787f9f20d3a01445f92dfc6845a3c898bd25f57/packages/studio-base/src/randomAccessDataProviders/MemoryCacheDataProvider.ts#L555-L564

This then ends up leading to the below being executed:

https://github.com/foxglove/studio/blob/0787f9f20d3a01445f92dfc6845a3c898bd25f57/packages/studio-base/src/randomAccessDataProviders/MemoryCacheDataProvider.ts#L324-L327

Depending on the value of `this._startTime` previously, this will result in `blockRange.end - blockRange.start` being either `1` or `2`. If it is `2`, then we end up dying inside of

https://github.com/foxglove/studio/blob/0787f9f20d3a01445f92dfc6845a3c898bd25f57/packages/studio-base/src/randomAccessDataProviders/MemoryCacheDataProvider.ts#L435

because `this._readAheadBlocks` is `1`, and `this._readRequests[0]?.blockRange` has a range of `2`, which leads to an error on this line:
 https://github.com/foxglove/studio/blob/0787f9f20d3a01445f92dfc6845a3c898bd25f57/packages/studio-base/src/util/getNewConnection.ts#L60

[image]

**Steps To Reproduce**
1. Open a logfile (I was using an ~8GB MCAP file with 10 MB chunks, that covered ~3000 seconds of data).
2. While the file is loading, click around the seek bar, with or without trying to play (I've observed this with the stack trace pointing to both calls to `_getMessages` in `RandomAccessPlayer.ts`).

This seems to be a non-issue if I let the entire file load before attempting to click the play/pause button or seek to a given location.

Additionally, once this error occurs, studio appears to stop trying to load any further pieces of the logfile.

defunctzombie, 2022-03-29:
@jameskuszmaul-brt Could you try enabling the "experimental bag player" in the App Settings and see if that is a workaround to the particular issue you are running into here?

jameskuszmaul-brt, 2022-03-29:
It seems to be already checked? Not sure if I did that or not, so will try un-checking that. Also already tried "unlimited in-memory cache", to no effect.

defunctzombie, 2022-03-29:
Oh sorry - I see this is mcap not bag file. That setting won't impact this.

closed 2022-04-19 by jtbandes