foxglove/studio #3100: Failed to execute 'setItem' on 'Storage': Setting the value of 'studio.layouts.local.7c34963d-56e9-41e9-add7-3788a35e1b30' exceeded the quota.

2metres, 2022-03-29:
**Description**
[image]

- Version: Dev
- OS: MacOS

**Steps To Reproduce**

Have a large number of layouts and load the example layout

**Expected Behavior**

**Actual Behavior**

jtbandes, 2022-03-30:
I think we might want to just switch to IndexedDB for the layout storage. While having lots of layouts is not ideal we shouldn't be imposing arbitrary limits on users adding lots of layouts if they really want to.

jtbandes, 2022-03-31:
> Why's it not ideal?

Because scrolling through a huge list is a bad UX. Maybe this is saying more about our layout UX than about the number of layouts.

https://web.dev/storage-for-the-web/#other

foxymiles, 2022-04-18:
Sounds like the strategy here is to switch to IndexedDB and do a one-time migration from localStorage?

defunctzombie, 2022-04-18:
> Sounds like the strategy here is to switch to IndexedDB and do a one-time migration from localStorage?

Yep.

closed 2022-04-21 by jtbandes